corecoding/Vitals #154: "No Data" for my btrfs drive.

cdysthe, 2020-06-28:
**Describe the bug**

I am getting "no data" for btrfs formatted drives/volumes.

**To Reproduce**
Steps to reproduce the behavior:
1. Point to a location like "/"
2. Check "Storage" in the extension drop-down
3. See "No Data" 

**Expected behavior**
Data for the path set, for instance "/"

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04 with default desktop

**Additional context**
I can not be sure whether this is related to my storage as such or the fact that is has btrfs volumes on it.

corecoding, 2020-06-30:
Did you follow the instructions here? https://github.com/corecoding/Vitals/wiki/Storage-reporting-%22No-Data%22

corecoding, 2020-07-08:
Closing due to no response

closed 2020-07-08 by corecoding