corecoding/Vitals #158: Support for HDD IOPS / bytes

easel, 2020-07-21:
**Describe the new feature you'd like**
I'd like to see the amount of hdd bytes read/written and read/write ops for each disk.

**Additional context**
IOTOP displays the information I'm asking for, but I don't where's it's read from and it requires sudo so I'm not fully optimistic it's just a /proc read away.

corecoding, 2020-08-24:
Is there a formula that you know of to calculate IOPS?

dorsi, 2021-11-02:
maybe iotop -ob | egrep "Current DISK READ|Current DISK WRITE"

davidbullado, 2023-08-11:
This command seems to work :+1: 

```
Current DISK READ:    1355.82 K/s | Current DISK WRITE:       0.00 B/s
Current DISK READ:     658.96 K/s | Current DISK WRITE:       0.00 B/s
Current DISK READ:     466.95 K/s | Current DISK WRITE:       0.00 B/s
Current DISK READ:      15.59 K/s | Current DISK WRITE:     311.79 K/s
```

It could be very interesting to implement it.