corecoding/Vitals #164: How to remove defunct monitors from the top bar

wolfen351, 2020-08-24:
**Describe the bug**
Some icons are in the top bar that I cannot remove:

[image]

The icons shown with the "..." next to them are probably legacy sensors or something, I'd like to clear them, but because they don't exist anymore I can't untick them in the menus

**To Reproduce**
Steps to reproduce the behavior: 

Not sure but I think this may be the way:
1. Have sensors in the pc
2. Select them for display
3. Remove the driver for those sensors
4. The items are displayed, but no way to clear them out.

**Expected behavior**
Perhaps hide the icons and items if there is no data

**Screenshots**
See above

**Desktop (please complete the following information):**
Ubuntu 20.04.1 LTS
 - Gnome Version 3.36.3

corecoding, 2020-08-24:
If you click any existing sensor, it should clear the missing ... ones. Please let me know either way if this works or not.

wolfen351, 2020-08-25:
It worked perfectly, thank you :+1:

closed 2020-08-25 by wolfen351