corecoding/Vitals #165: Error upgrading Vitalis

cristiklein, 2020-08-27:
**Describe the bug**

I recently upgraded Vitalis and got "ERROR" (see screenshot below).

`sudo journalctl /usr/bin/gnome-shell -f` reports:

```
aug 27 09:08:29 minitrue gnome-shell[3318]: JS ERROR: Extension Vitals@CoreCoding.com: Error: A different version was loaded previously. You need to log out for changes to take effect.
                                            loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:341:52
                                            callback@resource:///org/gnome/shell/ui/extensionDownloader.js:234:39
                                            gotExtensionZipFile/<@resource:///org/gnome/shell/ui/extensionDownloader.js:115:13
```

While the error seems harmless, it may easily confuse the user.

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure you have the previous version of Vitalis installed.
2. Wait for Gnome Shell to notify you that a new version of Vitalis is available.
3. Update Vitalis.
4. You will see the error in the attached screenshot.

**Expected behavior**
Either the update should succeed, or the user should get a message "Please logout for the update to take effect."

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Gnome Version 3.36.4

**Additional context**
Add any other context about the problem here.

corecoding, 2020-08-27:
The restart requirement is not of my doing. There is a bug somewhere higher up in Gnome Shell. I've personally never seen this before. My only theory is that perhaps someone had a higher version installed as specified in metadata.json under the "version" key. Perhaps on Manjaro who install it outside of Gnome Extensions, or were beta testers.

A few other extensions are having the same issue:
https://github.com/mzur/gnome-shell-wsmatrix/issues/118
https://github.com/ai/autohide-battery/issues/13

rvalle, 2020-09-06:
No! I just updated and it entered into error state too. Ubuntu 20.04, pretty recent installation.

I really like and use this extension, is it possible to roll back the update?

corecoding, 2020-09-06:
If I could be certain it would fix the issue, perhaps. This is not a code related issue from what I can tell. I would like to ask the community - is there a forum, or email address where we (I) can ask for help? I can't even find the files on my system mentioned in the stack trace or the gnome-shell repo.

rvalle, 2020-09-07:
@corecoding, I don't know how, but the extension is now working, and it is the last version.

I don't think I have rebooted the system. But I did restart gnome this morning (alt+f2, R). 

I don't know at which point it appeared back, but there it is. 

@cristiklein is it working for you? I think you are also on the same Ubuntu...

cristiklein, 2020-09-07:
I rebooted my computer and it worked like a charm. But yeah, it was
confusing at first.

carlocaione, 2020-10-12:
What helped for me, no idea if I was just lucky but maybe this can help others: if you are running on Wayland quit it and restart in Xorg. After that it starts working again on Wayland.

closed 2020-11-05 by corecoding

corecoding, 2020-11-05:
Just wanted to chime in with the exact reason this is happening. To recap, this is something out of my control:

https://gitlab.gnome.org/GNOME/gnome-shell/-/merge_requests/1248

ZaLiTHkA, 2020-11-06:
I came looking for a changelog to see what the latest update contained, which is when I found this issue mentioned in a recent commit.

FWIW, I saw the browser notification from Gnome Extensions, and I updated Vitals now this morning with no issues.. so @corecoding, in response to your `up version by two, trying to fix #165` commit message, it looks like it might indeed be resolved..?

corecoding, 2020-11-06:
It was not my "up version by two". If you look at your meta file, gnome
extensions actually decremented the version number on upload. I believe it
is a one time gnome bug, as mentioned in a link in the #165 ticket.

commit 9b5af41a (2020-11-05): up version by two, trying to fix #165