corecoding/Vitals #173: Error on Upgrade - Broken Install

SterlingButters, 2020-11-12:
**Describe the bug**
Recent upgrade results in gnome shell extension ERROR. Was working flawlessly before that. 

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt Upgrade/Install from https://extensions.gnome.org/extension/1460/vitals/

**Expected behavior**
Successful Upgrade/Installation

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
GNOME Shell 3.36.4

**Additional context**
Maybe I can purge remove the installation and try again? Not sure how this would be done though

corecoding, 2020-11-12:
I just created a Wiki entry for this issue. Please see https://github.com/corecoding/Vitals/wiki/Broken-after-upgrade

closed 2020-11-12 by corecoding

SterlingButters, 2020-11-12:
Chris,
  
  ​
  
  ​Thanks for such a quick response and resolution!