corecoding/Vitals #174: Vitals displaying wrong fan RPM values. Zero values never displayed.

antonvalletas, 2020-11-14:
**Describe the bug**
Vitals display fan RPM around 2400 when both fans are actually off, as seen by putting the laptop to your ear and/or using the sensors command. 

**To Reproduce**
Steps to reproduce the behavior:
1. Update to latest version on extensions.gnome.org
2. Same config that was working on last update, now never displays 0rpm fan speeds, it just stays stuck at the last non-zero rpm value. 

**Expected behavior**
Vitals should display same fan rpm value as sensors. 

**Screenshots**
[https://imgur.com/HFVyTBc](url)
[https://imgur.com/3rCk3gf](url)

**Desktop (please complete the following information):**
 - OS: Linux xps 5.9.8-arch1-1 #1 SMP PREEMPT Tue, 10 Nov 2020 22:44:11 +0000 x86_64 GNU/Linux
 - Gnome Version:  3.38.1

**Additional context**
Only started happening after updating from extensions.gnome.org a few days ago. It's only happening on the latest version available on extensions.gnome.org, previous versions were unaffected.

theodoreb, 2020-11-14:
Same here, probably related to #172 and #149 this makes the value useless for me. 

I'm not sure it's relevant to have an average fan speed too.

antonvalletas, 2020-11-14:
In some cases it might be a good measure. For example, in the case of the XPS 9570, both fans spin at slightly different speeds, with just a difference of 10-100RPM, i.e: when one fan is 2435, the other might be 2467, so averaging those two measures out might be a nice feature, however right now the fan speed measure is in fact broken. Sometimes displaying nodata and other times displaying 2463rpm when both fans are in fact completely off.

AndreaTosti, 2020-11-17:
Same issue:

```
Linux workstation 5.8.0-7630-generic #32~1605108806~20.10~7e52b13-Ubuntu SMP Wed Nov 11 19:10:30 UTC  x86_64 x86_64 x86_64 GNU/Linux
Gnome Version: 3.38.1
Vitals version: 38
```

`sensors` output:

```
tosti@workstation:~$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +36.0°C  (high = +100.0°C, crit = +100.0°C)

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +38.0°C  

BAT0-acpi-0
Adapter: ACPI interface
in0:          15.56 V  

thinkpad-isa-0000
Adapter: ISA adapter
fan1:           0 RPM
temp1:        +38.0°C  
```

[image]

corecoding, 2020-11-17:
I agree that #172 is the cause. I will fix and upload an update tonight. It could either be immediately approved, or take up to a week, depends on how active the extension reviewers are. For those who want a fix now, as mentioned in the commit, you can edit the file locally until the update is released.

corecoding, 2020-11-19:
I have pushed a simple code block comment to undo #172 into the develop branch. Can one of you confirm if the issue is fixed?

AndreaTosti, 2020-11-29:
> I have pushed a simple code block comment to undo #172 into the develop branch. Can one of you confirm if the issue is fixed?

I did a quick test (played a 4K video to stress my cpu), it started at 0 RPM, then ramped up and eventually went back to 0 rpm. 
If I have to test anything else let me know.

corecoding, 2020-12-01:
Thank you! I have submitted the update to the extension store. I'm going to close the ticket, please let me know if you still have issues after the update. Since you did the manual git clone, you may have to uninstall and re-install to get the newest version.

closed 2020-12-01 by corecoding