corecoding/Vitals #175: Viitals crashing #2

heynnema, 2020-11-14:
```
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: == Stack trace for context 0x5650a1ddb6e0 ==
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #0   5650cddeba58 i   /home/alheynneman/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:507 (2b490f4dd970 @ 90)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #1   7ffeba08cbb0 b   resource:///org/gnome/gjs/modules/script/_legacy.js:82 (370afa5a9970 @ 69)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #2   5650cddeb9d0 i   /home/alheynneman/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:535 (2b490f4d18f8 @ 12)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #3   5650cddeb8f8 i   resource:///org/gnome/shell/ui/extensionSystem.js:108 (1d3580764628 @ 394)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #4   5650cddeb868 i   resource:///org/gnome/shell/ui/extensionSystem.js:624 (1d358076a8f8 @ 15)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #5   7ffeba08dae0 b   self-hosted:266 (370afa5ace20 @ 259)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #6   5650cddeb7d8 i   resource:///org/gnome/shell/ui/extensionSystem.js:623 (1d358076a880 @ 98)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #7   5650cddeb748 i   resource:///org/gnome/shell/ui/extensionSystem.js:641 (1d358076a970 @ 82)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #8   7ffeba08e9a0 b   self-hosted:1009 (140f819a2268 @ 423)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #9   7ffeba08eaa0 b   resource:///org/gnome/gjs/modules/core/_signals.js:133 (370afa5ac880 @ 427)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #10   7ffeba08f780 b   resource:///org/gnome/shell/ui/sessionMode.js:198 (140f81925880 @ 286)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #11   5650cddeb4e8 i   resource:///org/gnome/shell/ui/sessionMode.js:159 (140f819256a0 @ 40)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #12   5650cddeb430 i   resource:///org/gnome/shell/ui/screenShield.js:574 (140f81913880 @ 191)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #13   5650cddeb398 i   resource:///org/gnome/shell/ui/screenShield.js:311 (140f8190ed30 @ 36)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #14   7ffeba090460 b   self-hosted:1011 (140f819a2268 @ 454)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #15   5650cddeb308 i   resource:///org/gnome/shell/ui/lightbox.js:195 (1d3580733d30 @ 29)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #16   5650cddeb278 i   resource:///org/gnome/shell/ui/environment.js:75 (370afa58d5b0 @ 98)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #17   5650cddeb1e8 i   resource:///org/gnome/shell/ui/environment.js:149 (370afa58d9e8 @ 14)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: == Stack trace for context 0x5650a1ddb6e0 ==
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #0   5650cddeba58 i   /home/alheynneman/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:507 (2b490f4dd970 @ 90)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #1   7ffeba08cbb0 b   resource:///org/gnome/gjs/modules/script/_legacy.js:82 (370afa5a9970 @ 69)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x5650c2dfd760), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #2   5650cddeb9d0 i   /home/alheynneman/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:535 (2b490f4d18f8 @ 12)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #3   5650cddeb8f8 i   resource:///org/gnome/shell/ui/extensionSystem.js:108 (1d3580764628 @ 394)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #4   5650cddeb868 i   resource:///org/gnome/shell/ui/extensionSystem.js:624 (1d358076a8f8 @ 15)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #5   7ffeba08dae0 b   self-hosted:266 (370afa5ace20 @ 259)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #6   5650cddeb7d8 i   resource:///org/gnome/shell/ui/extensionSystem.js:623 (1d358076a880 @ 98)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #7   5650cddeb748 i   resource:///org/gnome/shell/ui/extensionSystem.js:641 (1d358076a970 @ 82)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #8   7ffeba08e9a0 b   self-hosted:1009 (140f819a2268 @ 423)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #9   7ffeba08eaa0 b   resource:///org/gnome/gjs/modules/core/_signals.js:133 (370afa5ac880 @ 427)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #10   7ffeba08f780 b   resource:///org/gnome/shell/ui/sessionMode.js:198 (140f81925880 @ 286)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #11   5650cddeb4e8 i   resource:///org/gnome/shell/ui/sessionMode.js:159 (140f819256a0 @ 40)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #12   5650cddeb430 i   resource:///org/gnome/shell/ui/screenShield.js:574 (140f81913880 @ 191)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #13   5650cddeb398 i   resource:///org/gnome/shell/ui/screenShield.js:311 (140f8190ed30 @ 36)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #14   7ffeba090460 b   self-hosted:1011 (140f819a2268 @ 454)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #15   5650cddeb308 i   resource:///org/gnome/shell/ui/lightbox.js:195 (1d3580733d30 @ 29)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #16   5650cddeb278 i   resource:///org/gnome/shell/ui/environment.js:75 (370afa58d5b0 @ 98)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #17   5650cddeb1e8 i   resource:///org/gnome/shell/ui/environment.js:149 (370afa58d9e8 @ 14)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: == Stack trace for context 0x5650a1ddb6e0 ==
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #0   5650cddeba58 i   /home/alheynneman/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:507 (2b490f4dd970 @ 90)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #1   7ffeba08cbb0 b   resource:///org/gnome/gjs/modules/script/_legacy.js:82 (370afa5a9970 @ 69)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #2   5650cddeb9d0 i   /home/alheynneman/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:535 (2b490f4d18f8 @ 12)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #3   5650cddeb8f8 i   resource:///org/gnome/shell/ui/extensionSystem.js:108 (1d3580764628 @ 394)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #4   5650cddeb868 i   resource:///org/gnome/shell/ui/extensionSystem.js:624 (1d358076a8f8 @ 15)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #5   7ffeba08dae0 b   self-hosted:266 (370afa5ace20 @ 259)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #6   5650cddeb7d8 i   resource:///org/gnome/shell/ui/extensionSystem.js:623 (1d358076a880 @ 98)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #7   5650cddeb748 i   resource:///org/gnome/shell/ui/extensionSystem.js:641 (1d358076a970 @ 82)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #8   7ffeba08e9a0 b   self-hosted:1009 (140f819a2268 @ 423)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #9   7ffeba08eaa0 b   resource:///org/gnome/gjs/modules/core/_signals.js:133 (370afa5ac880 @ 427)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #10   7ffeba08f780 b   resource:///org/gnome/shell/ui/sessionMode.js:198 (140f81925880 @ 286)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #11   5650cddeb4e8 i   resource:///org/gnome/shell/ui/sessionMode.js:159 (140f819256a0 @ 40)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #12   5650cddeb430 i   resource:///org/gnome/shell/ui/screenShield.js:574 (140f81913880 @ 191)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #13   5650cddeb398 i   resource:///org/gnome/shell/ui/screenShield.js:311 (140f8190ed30 @ 36)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #14   7ffeba090460 b   self-hosted:1011 (140f819a2268 @ 454)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #15   5650cddeb308 i   resource:///org/gnome/shell/ui/lightbox.js:195 (1d3580733d30 @ 29)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #16   5650cddeb278 i   resource:///org/gnome/shell/ui/environment.js:75 (370afa58d5b0 @ 98)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: #17   5650cddeb1e8 i   resource:///org/gnome/shell/ui/environment.js:149 (370afa58d9e8 @ 14)
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: == Stack trace for context 0x5650a1ddb6e0 ==
[... cut ...]
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x5650c46a7260), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
Nov 13 07:41:41 Satellite-E55 gnome-shell[3122]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
```

corecoding, 2020-11-19:
Would you mind testing the develop branch to see if the issue is fixed? I personally don't see these dealloc messages.

corecoding, 2020-12-01:
I am closing this issue because I haven't received a comment. I believe this issue to be fixed. Please let me know once the update posts if it you are still having issues.

closed 2020-12-01 by corecoding

heynnema, 2020-12-05:
Sorry for the long delay. The crashing is still occurring with the latest release. Let me know if you need more info.

commit 9d85638d (2020-11-19): attempt to fix #175