corecoding/Vitals #176: Hint on adding new sensors to panel

qm2k, 2020-11-26:
It took me a while to find how to add new sensors to panel, and I know what github is. Guess many Gnome users won't even bother.

A static line like "Clicking on a sensor adds it to the panel" in the sensors list would be helpful in this regard.

0x427279, 2021-08-14:
Thanks for raising this issue @qm2k, I've been searching for how to add sensors to the panel, and would have probably given up and looked for an alternative but for this issue describing how it can be done. It seems obvious now I know, but the fact I had to search for how to do it demonstrates it's easily missed. Now that I've figured it out, I have a new favourite extension, but I also need a wider monitor for all the sensors I've added :)

@corecoding maybe also worth a mention in the readme.md?

corecoding, 2022-02-20:
This is part of the gnome extension "store" and I haven't seen anyone ask the question since. Closing, than you!

closed 2022-02-20 by corecoding