corecoding/Vitals #177: Vitals crashing #3

heynnema, 2020-12-05:
Ubuntu 20.10

Latest release Vitals.

Please see previous #171 and #175 

This crash repeats itself over and over.

```
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55ddfa8dd490), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: == Stack trace for context 0x55dd62b55220 ==
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #0   55de02a77be8 i   /home/alheynneman/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:507 (15cf45957ab0 @ 97)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #1   7ffdb8b73480 b   resource:///org/gnome/gjs/modules/script/_legacy.js:82 (47aa5b991a0 @ 69)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #2   55de02a77b60 i   /home/alheynneman/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:535 (15cf45938f10 @ 12)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #3   55de02a77a88 i   resource:///org/gnome/shell/ui/extensionSystem.js:109 (15cf45938c90 @ 395)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #4   55de02a779f8 i   resource:///org/gnome/shell/ui/extensionSystem.js:620 (15cf459380b0 @ 15)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #5   7ffdb8b74100 b   self-hosted:225 (47aa5b99a60 @ 273)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #6   55de02a77968 i   resource:///org/gnome/shell/ui/extensionSystem.js:619 (15cf45938100 @ 98)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #7   55de02a778d8 i   resource:///org/gnome/shell/ui/extensionSystem.js:637 (15cf45938060 @ 82)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #8   7ffdb8b74c60 b   self-hosted:850 (37514ac06510 @ 423)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #9   7ffdb8b74d50 b   resource:///org/gnome/gjs/modules/core/_signals.js:133 (47aa5b995b0 @ 439)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #10   7ffdb8b759d0 b   resource:///org/gnome/shell/ui/sessionMode.js:198 (3025c579fbf0 @ 284)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #11   55de02a77678 i   resource:///org/gnome/shell/ui/sessionMode.js:159 (3025c579fd30 @ 40)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #12   55de02a775c0 i   resource:///org/gnome/shell/ui/screenShield.js:575 (3025c578bb00 @ 208)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #13   55de02a77528 i   resource:///org/gnome/shell/ui/screenShield.js:311 (3025c5792290 @ 36)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #14   7ffdb8b76640 b   self-hosted:850 (37514ac06510 @ 454)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #15   55de02a77498 i   resource:///org/gnome/shell/ui/lightbox.js:193 (15cf4590a290 @ 29)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #16   55de02a77408 i   resource:///org/gnome/shell/ui/environment.js:84 (47aa5b84bf0 @ 98)
Dec  3 07:28:58 Satellite-E55 gnome-shell[264280]: #17   55de02a77378 i   resource:///org/gnome/shell/ui/environment.js:165 (47aa5b848d0 @ 14)
```

corecoding, 2020-12-05:
You are using the develop branch right? The latest release has not been released yet on Gnome Extensions. You are likely still on the old version. https://extensions.gnome.org/review/

corecoding, 2020-12-05:
Also, how is #171 different from #177?

heynnema, 2020-12-05:
Sorry, I thought that the develop version had been released as a final. I just downloaded the development .zip file and unzipped it into my ~/.local/share/gnome-shell/extensions, restarted gnome-shell, and I don't see Vitals in my https://extensions.gnome.org/local/. What did I do wrong?

corecoding, 2020-12-05:
There are instructions over at https://github.com/corecoding/Vitals under the Beta Testing section. Don't forget to logout/login (or Alt-F2 r) to restart your shell. Also pay attention to your logs dates, making sure you aren't seeing errors from the current installed version.

BTW, you probably don't want to keep your system this way after you see if my fix worked, as the develop branch can be unstable at times. So, once you test the fix, you can remove the directory and re-install the extension from the portal.

heynnema, 2020-12-05:
Develop version installed. Is there a place where I can see the current version # of the extension?

heynnema, 2020-12-05:
metadata.json says version 39

corecoding, 2020-12-05:
Unfortunately no. There are no notable feature changes either. But if you logged out and back in, and the directory mentioned in the readme is indeed from GitHub (you can type git status to make sure) then what you are seeing on the desktop should be from the develop branch

heynnema, 2020-12-06:
Same problem...

```
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: == Stack trace for context 0x555e93739290 ==
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #0   7ffee1a8f1f0 b   /home/alheynneman/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:508 (3a3fe9e39f10 @ 128)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #1   7ffee1a8fda0 b   resource:///org/gnome/gjs/modules/script/_legacy.js:82 (b29b26991a0 @ 69)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #2   555ebd951a10 i   /home/alheynneman/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:536 (b29b26e10b0 @ 12)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #3   555ebd951938 i   resource:///org/gnome/shell/ui/extensionSystem.js:109 (3a3fe9e39c90 @ 395)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #4   555ebd9518a8 i   resource:///org/gnome/shell/ui/extensionSystem.js:620 (3a3fe9e390b0 @ 15)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #5   7ffee1a90a20 b   self-hosted:225 (b29b2699a60 @ 273)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #6   555ebd951818 i   resource:///org/gnome/shell/ui/extensionSystem.js:619 (3a3fe9e39100 @ 98)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #7   555ebd951788 i   resource:///org/gnome/shell/ui/extensionSystem.js:637 (3a3fe9e39060 @ 82)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #8   7ffee1a91580 b   self-hosted:850 (307771207510 @ 423)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #9   7ffee1a91670 b   resource:///org/gnome/gjs/modules/core/_signals.js:133 (b29b26995b0 @ 439)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #10   7ffee1a91e10 b   resource:///org/gnome/shell/ui/sessionMode.js:198 (36497169fbf0 @ 284)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #11   555ebd951628 i   resource:///org/gnome/shell/ui/sessionMode.js:159 (36497169fd30 @ 40)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #12   555ebd951570 i   resource:///org/gnome/shell/ui/screenShield.js:575 (36497168bb00 @ 208)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #13   555ebd9514d8 i   resource:///org/gnome/shell/ui/screenShield.js:311 (364971693290 @ 36)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #14   7ffee1a92a80 b   self-hosted:850 (307771207510 @ 454)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #15   555ebd951448 i   resource:///org/gnome/shell/ui/lightbox.js:193 (3a3fe9e0a290 @ 29)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #16   555ebd9513b8 i   resource:///org/gnome/shell/ui/environment.js:84 (b29b2684bf0 @ 98)
Dec  5 17:25:56 Satellite-E55 gnome-shell[2934]: #17   555ebd951328 i   resource:///org/gnome/shell/ui/environment.js:165 (b29b26848d0 @ 14)
```

corecoding, 2020-12-11:
I am unable to duplicate the issue, so it is hard to fix. Does anyone else out there have this issue?

krashish8, 2021-01-13:
@corecoding Yes, I have the same issue on `Ubuntu 20.04` on `GNOME Shell 3.36.4`. I installed the extension from https://extensions.gnome.org/

<details>
<summary><code>Entire Error Logs</code></summary>
<pre>
Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55f218ef8480), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
== Stack trace for context 0x55f20de68710 ==
#0   55f218fdcde8 i   /home/ashish/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:508 (213469376b50 @ 121)
#1   7fff81607b00 I   resource:///org/gnome/gjs/modules/script/_legacy.js:82 (f3e601a9970 @ 69)
#2   55f218fdcd60 i   /home/ashish/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:536 (f3e601b2088 @ 12)
#3   55f218fdcc88 i   resource:///org/gnome/shell/ui/extensionSystem.js:108 (27439ff64628 @ 394)
#4   55f218fdcbf8 i   resource:///org/gnome/shell/ui/extensionSystem.js:505 (27439ff6a2e0 @ 15)
#5   7fff81608a30 b   self-hosted:266 (f3e601ace20 @ 259)
#6   55f218fdcb68 i   resource:///org/gnome/shell/ui/extensionSystem.js:504 (27439ff6a100 @ 120)
#7   7fff81609900 b   self-hosted:1011 (3bbccc0a1268 @ 454)
#8   7fff8160ba40 I   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:529 (f3e6019c970 @ 169)
#9   55f218fdcab8 i   resource:///org/gnome/shell/ui/extensionSystem.js:209 (27439ff64808 @ 200)
#10   55f218fdca28 i   resource:///org/gnome/shell/ui/shellDBus.js:311 (3bbccc015e20 @ 20)
#11   55f218fdc940 i   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:327 (f3e6019c1f0 @ 185)
#12   55f218fdc888 i   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:404 (f3e6019c3d0 @ 34)
== Stack trace for context 0x55f20de68710 ==
#0   55f218fdcde8 i   /home/ashish/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:508 (213469376b50 @ 121)
#1   7fff81607b00 I   resource:///org/gnome/gjs/modules/script/_legacy.js:82 (f3e601a9970 @ 69)
#2   55f218fdcd60 i   /home/ashish/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:536 (f3e601b2088 @ 12)
#3   55f218fdcc88 i   resource:///org/gnome/shell/ui/extensionSystem.js:108 (27439ff64628 @ 394)
#4   55f218fdcbf8 i   resource:///org/gnome/shell/ui/extensionSystem.js:505 (27439ff6a2e0 @ 15)
#5   7fff81608a30 b   self-hosted:266 (f3e601ace20 @ 259)
#6   55f218fdcb68 i   resource:///org/gnome/shell/ui/extensionSystem.js:504 (27439ff6a100 @ 120)
#7   7fff81609900 b   self-hosted:1011 (3bbccc0a1268 @ 454)
#8   7fff8160ba40 I   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:529 (f3e6019c970 @ 169)
#9   55f218fdcab8 i   resource:///org/gnome/shell/ui/extensionSystem.js:209 (27439ff64808 @ 200)
#10   55f218fdca28 i   resource:///org/gnome/shell/ui/shellDBus.js:311 (3bbccc015e20 @ 20)
#11   55f218fdc940 i   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:327 (f3e6019c1f0 @ 185)
#12   55f218fdc888 i   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:404 (f3e6019c3d0 @ 34)
== Stack trace for context 0x55f20de68710 ==
#0   55f218fdcde8 i   /home/ashish/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:508 (213469376b50 @ 121)
#1   7fff81607b00 I   resource:///org/gnome/gjs/modules/script/_legacy.js:82 (f3e601a9970 @ 69)
#2   55f218fdcd60 i   /home/ashish/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:536 (f3e601b2088 @ 12)
#3   55f218fdcc88 i   resource:///org/gnome/shell/ui/extensionSystem.js:108 (27439ff64628 @ 394)
#4   55f218fdcbf8 i   resource:///org/gnome/shell/ui/extensionSystem.js:505 (27439ff6a2e0 @ 15)
#5   7fff81608a30 b   self-hosted:266 (f3e601ace20 @ 259)
#6   55f218fdcb68 i   resource:///org/gnome/shell/ui/extensionSystem.js:504 (27439ff6a100 @ 120)
clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
#7   7fff81609900 b   self-hosted:1011 (3bbccc0a1268 @ 454)
#8   7fff8160ba40 I   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:529 (f3e6019c970 @ 169)
#9   55f218fdcab8 i   resource:///org/gnome/shell/ui/extensionSystem.js:209 (27439ff64808 @ 200)
#10   55f218fdca28 i   resource:///org/gnome/shell/ui/shellDBus.js:311 (3bbccc015e20 @ 20)
#11   55f218fdc940 i   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:327 (f3e6019c1f0 @ 185)
#12   55f218fdc888 i   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:404 (f3e6019c3d0 @ 34)
== Stack trace for context 0x55f20de68710 ==
#0   55f218fdcde8 i   /home/ashish/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:508 (213469376b50 @ 121)
#1   7fff81607b00 I   resource:///org/gnome/gjs/modules/script/_legacy.js:82 (f3e601a9970 @ 69)
#2   55f218fdcd60 i   /home/ashish/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:536 (f3e601b2088 @ 12)
#3   55f218fdcc88 i   resource:///org/gnome/shell/ui/extensionSystem.js:108 (27439ff64628 @ 394)
#4   55f218fdcbf8 i   resource:///org/gnome/shell/ui/extensionSystem.js:505 (27439ff6a2e0 @ 15)
#5   7fff81608a30 b   self-hosted:266 (f3e601ace20 @ 259)
#6   55f218fdcb68 i   resource:///org/gnome/shell/ui/extensionSystem.js:504 (27439ff6a100 @ 120)
#7   7fff81609900 b   self-hosted:1011 (3bbccc0a1268 @ 454)
#8   7fff8160ba40 I   resource:///org/gnome/gjs/modules/core/overrides/Gio.js:529 (f3e6019c970 @ 169)
#9   55f218fdcab8 i   resource:///org/gnome/shell/ui/extensionSystem.js:209 (27439ff64808 @ 200)
[... cut ...]
Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55f20f158860), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
</pre>
</details>

For me, this happens when I disable this extension. So, there is an error in the `disable()` function, during the deallocation of the menu items. From the error logs, I guess that probably `.destroy()` is called on an object which is already deallocated.

I generated the error logs from `journalctl /usr/bin/gnome-shell -f -o cat`

corecoding, 2021-07-12:
This has been fixed in the Gnome 40 version of Vitals. This means that if you are using Gnome 3.38 or lower, you will still see the error. Can anyone who is using Gnome 40 try out the master branch? (It is not in the extension store yet)

corecoding, 2021-07-13:
This fix was release for both v3.38 and v40. Will open if you see the same issue. If you see a different issue please open another ticket.

closed 2021-07-13 by corecoding