corecoding/Vitals #178: No Data shown in Fan section on Latitude 5501 even though the fan is visible in /sys and sensors output

qm2k, 2020-12-07:
**Describe the bug**
On a Dell Latitude 5501 with i8kutils installed the fan information is present in `sensors` command output and in `/sys/devices/virtual/hwmon/hwmon3/` files, but missing in the Vitals drop-down (there's `No Data` in the `Fan` line and the submenu does not open). The Vitals extension was installed from git, I'm on commit 9b5af41a57e82e6df243c2986eb52f8379272bfd. Ubuntu 18.04, Gnome 3.28.2 in case that matters.

```
(master) ~/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +98.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +97.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +95.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +98.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +81.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +96.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +93.0°C  (high = +100.0°C, crit = +100.0°C)

dell_smm-virtual-0
Adapter: Virtual device
fan1:        3383 RPM

(master) ~/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com$ cat /sys/devices/virtual/hwmon/hwmon3/pwm1 
128
(master) ~/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com$ find /sys/devices/virtual/hwmon/hwmon3/
/sys/devices/virtual/hwmon/hwmon3/
/sys/devices/virtual/hwmon/hwmon3/uevent
/sys/devices/virtual/hwmon/hwmon3/fan1_label
/sys/devices/virtual/hwmon/hwmon3/power
/sys/devices/virtual/hwmon/hwmon3/power/runtime_active_time
/sys/devices/virtual/hwmon/hwmon3/power/runtime_active_kids
/sys/devices/virtual/hwmon/hwmon3/power/runtime_usage
/sys/devices/virtual/hwmon/hwmon3/power/runtime_status
/sys/devices/virtual/hwmon/hwmon3/power/autosuspend_delay_ms
/sys/devices/virtual/hwmon/hwmon3/power/async
/sys/devices/virtual/hwmon/hwmon3/power/runtime_suspended_time
/sys/devices/virtual/hwmon/hwmon3/power/runtime_enabled
/sys/devices/virtual/hwmon/hwmon3/power/control
/sys/devices/virtual/hwmon/hwmon3/pwm1
/sys/devices/virtual/hwmon/hwmon3/fan1_input
/sys/devices/virtual/hwmon/hwmon3/subsystem
/sys/devices/virtual/hwmon/hwmon3/name
(master) ~/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
(master) ~/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com$ git log
commit 9b5af41a57e82e6df243c2986eb52f8379272bfd (HEAD -> master, origin/master, origin/HEAD)
Author: Chris Monahan <chris@corecoding.com>
Date:   Thu Nov 5 06:55:39 2020 -0800

    up version by two, trying to fix #165 
```

corecoding, 2020-12-11:
Did you see this page? https://github.com/corecoding/Vitals/wiki/Missing-voltage-or-fan-data

qm2k, 2020-12-11:
> Did you see this page? https://github.com/corecoding/Vitals/wiki/Missing-voltage-or-fan-data

Yes, I did. RPM is not zero, and something in /sys directory exists, so I thought you might be interested. I'm not missing the fan indication too much (unfortunately, it is loud enough to be heard and speed recognized by tone), so as for me you may consider this a low priority issue.

qm2k, 2020-12-11:
Suddenly, it worked! I have no idea what has changed, but I have updated both system and bios at least once.

closed 2020-12-11 by qm2k