corecoding/Vitals #179: Stops Working / Doesn't Work with 5950x CPU ?

lordbuddha, 2020-12-10:
**Describe the bug**
It was working.  I swapped my CPU from 2700X to the 5950X.    After that swap, Vitals stopped functioning.   

In panel it just displays a triangle, it the setting it displays correct motherboard/CPU and 32 CPUs.  But there is no frequency displayed for any CPU.  There are no governors.

I tried the ALT-F2 r sequence and logged out / logged in. 

**To Reproduce**
Steps to reproduce the behavior:
1. Change to 5950X CPU

**Expected behavior**
It to be functional ;)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Ubuntu 20.04
 - Gnome Version whatever is latest that comes [with](url) 20.04

**Additional context**
Add any other context about the problem here.

corecoding, 2020-12-10:
Can you share a screenshot of Vitals? I don't know what this screenshot is, but it is certainly not Vitals. :)

lordbuddha, 2020-12-10:
Wrong project then.   Please close.

closed 2020-12-11 by corecoding