corecoding/Vitals #180: v39 Ryzen vCore reporting without 2x multiplier

adamfmatt, 2021-01-16:
OS: Debian GNU/Linux bullseye/sid x86_64
Kernel: 5.10.0-1-amd64
DE: GNOME 3.38.2 
CPU: AMD Ryzen 5 2400G (8)
Vitals: version 39

Since the last kernel update 21/1/11(maybe irrelevant, was a large update before Bullseye soft freeze) - apt/history.log avalible upon request - Vitals is reporting vCore(in0) as it's unmodified value.

Vitals -     0.4V
sensors - 688.00 mV

sensors.conf for nct6792-isa-0290
label   in0     "Vcore"
   compute in0     @*2, @/2
   set     in0_min 0.16
   set     in0_max 1.46
   set     in0_beep 1

Small inconvince really perhaps something obvious I have missed? Great widget! Thank you Vitals team!

corecoding, 2022-02-20:
is there a file /in /sys/class/hwmon that can tell me if there is a multiplier being used for voltage?

adamfmatt, 2022-02-20:
Hi nice to hear from you.

I first want to say that I am in no way an expert.

As far as I know, apps like `lm-sensors` have to read the raw values from hwmon, then apply any math via a `sensors.conf` file. It seems these sensor monitoring chips ie the nct6792 on my board can vary widely as to their configuration. See https://github.com/lm-sensors/lm-sensors/tree/master/configs

Some of these config files contain educated guesses by those who wrote them and are in no way official.

It looks like a mess and wouldn't expect you to be able to take this into account.

I have notice in Vitals the registered values for "SVI2_Core" and "SVI2_SOC" are in fact correct.

I am not very concerned with this issue personally, feel free to close the issue at your discretion.

Thank you maintaining Vitals, I do enjoy it.

corecoding, 2022-02-22:
I will definitely work on this. There are other issues (tickets) talking about sensors.conf being used for alerting purposes. This will work well in reading this file for these multipliers.