corecoding/Vitals #181: Allow selection of processor speed units.

ishusain, 2021-01-25:
As a processor varies its speed, the measurement unit oscillates between MHz and GHz, causing that the icons shown in the top bar move constantly to adapt to the new size...E.g. 800.2MHz then 1.1GHZ  back to 700.5 MHz.
It would be great if we could set a fixed amount of significant digits and the units to use for each HW indicator.
Thanks, great job, a very usefull tool!

corecoding, 2021-01-25:
Does the fixed width setting under preferences work for you?

ishusain, 2021-01-26:
> Does the fixed width setting under preferences work for you?

Oh, I'm sorry, I didn't see it. Its working, but when the processor speed is below 1GHz, the "Hz" flickers a little bit. Something like:
1.1GHz -> 800.3M -> 800.3MHz -> 800.3M.

corecoding, 2022-05-05:
This appears to be a duplicate of #168. This feature is gaining traction so please chime in on #168 with your thumbs up etc

closed 2022-05-05 by corecoding