corecoding/Vitals #182: Drive says "No Data" whereas smartctl and nvme CLI are both OK

naholyr, 2021-01-26:
Popup shows no information about hard drive (nvme), it just says "No Data" and I have no temperature showing either:

[image]

- System is **Ubuntu 20.04.1 LTS** with **Gnome 3.36.8** using X11
- Computer is **Dell XPS 13 9370**
- Hard drive is **Samsung SSD 970 EVO Plus**
- `nvme` and `smartctl` both produce correct output, as `lsblk` correctly lists the disk (see below)
- I see no log anywhere, I tried to search for `Vitals` or `vitals` in `sudo journalctl /usr/bin/gnome-shell` but no luck, so I'm not sure if it's saying something

Here are commands outputs:

<details>
<summary><strong><code>lsblk</code></strong></summary>

```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
[…]
mmcblk0     179:0    0    29G  0 disk 
└─mmcblk0p1 179:1    0    29G  0 part /media/nchambrier/Live
nvme0n1     259:0    0 931,5G  0 disk 
├─nvme0n1p1 259:1    0   650M  0 part /boot/efi
├─nvme0n1p2 259:2    0   128M  0 part 
├─nvme0n1p3 259:3    0 169,4G  0 part 
├─nvme0n1p4 259:4    0   811M  0 part 
├─nvme0n1p5 259:5    0  11,9G  0 part 
├─nvme0n1p6 259:6    0   1,1G  0 part 
├─nvme0n1p7 259:7    0    86G  0 part /
├─nvme0n1p8 259:8    0 627,6G  0 part /home
└─nvme0n1p9 259:9    0    34G  0 part [SWAP]
```
</details>

<details>
<summary><strong><code>sudo nvme smart-log /dev/nvme0n1</code></strong></summary>

```
Smart Log for NVME device:nvme0n1 namespace-id:ffffffff
critical_warning                    : 0
temperature                         : 51 C
available_spare                     : 100%
available_spare_threshold           : 10%
percentage_used                     : 0%
data_units_read                     : 785 533
data_units_written                  : 2 857 632
host_read_commands                  : 22 256 893
host_write_commands                 : 67 909 152
controller_busy_time                : 830
power_cycles                        : 121
power_on_hours                      : 256
unsafe_shutdowns                    : 6
media_errors                        : 0
num_err_log_entries                 : 30
Warning Temperature Time            : 0
Critical Composite Temperature Time : 0
Temperature Sensor 1                : 51 C
Temperature Sensor 2                : 48 C
Thermal Management T1 Trans Count   : 0
Thermal Management T2 Trans Count   : 0
Thermal Management T1 Total Time    : 0
Thermal Management T2 Total Time    : 0
```
</details>

<details>
<summary><strong><code>sudo smartctl -a /dev/nvme0n1</code></strong></summary>

```
smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.4.0-64-generic] (local build)
Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       Samsung SSD 970 EVO Plus 1TB
Serial Number:                      [HIDDEN FROM COPY/PASTE]
Firmware Version:                   2B2QEXM7
PCI Vendor/Subsystem ID:            0x144d
IEEE OUI Identifier:                [HIDDEN FROM COPY/PASTE]
Total NVM Capacity:                 1 000 204 886 016 [1,00 TB]
Unallocated NVM Capacity:           0
Controller ID:                      4
Number of Namespaces:               1
Namespace 1 Size/Capacity:          1 000 204 886 016 [1,00 TB]
Namespace 1 Utilization:            481 041 756 160 [481 GB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            002538 5a01b064e8
Local Time is:                      Tue Jan 26 11:53:24 2021 CET
Firmware Updates (0x16):            3 Slots, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL Self_Test
Optional NVM Commands (0x005f):     Comp Wr_Unc DS_Mngmt Wr_Zero Sav/Sel_Feat Timestmp
Maximum Data Transfer Size:         512 Pages
Warning  Comp. Temp. Threshold:     85 Celsius
Critical Comp. Temp. Threshold:     85 Celsius

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     7.80W       -        -    0  0  0  0        0       0
 1 +     6.00W       -        -    1  1  1  1        0       0
 2 +     3.40W       -        -    2  2  2  2        0       0
 3 -   0.0700W       -        -    3  3  3  3      210    1200
 4 -   0.0100W       -        -    4  4  4  4     2000    8000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         0

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        51 Celsius
Available Spare:                    100%
Available Spare Threshold:          10%
Percentage Used:                    0%
Data Units Read:                    785 533 [402 GB]
Data Units Written:                 2 857 663 [1,46 TB]
Host Read Commands:                 22 256 894
Host Write Commands:                67 909 211
Controller Busy Time:               830
Power Cycles:                       121
Power On Hours:                     256
Unsafe Shutdowns:                   6
Media and Data Integrity Errors:    0
Error Information Log Entries:      30
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Temperature Sensor 1:               51 Celsius
Temperature Sensor 2:               53 Celsius

Error Information (NVMe Log 0x01, max 64 entries)
No Errors Logged
```
</details>

corecoding, 2021-01-26:
Vitals doesn't look at or use those commands. It reads data from /proc and /sys. Any output via the sensors command should be available/visible in Vitals. If your hard drive data is not in /proc or /sys, it will not show in Vitals.

naholyr, 2021-01-31:
How can I know this and why wouldn't it be lister in there? It's a classic nvme SSD, is there something I can do to debug or even fix this?

corecoding, 2021-03-18:
The kernel is responsible for populating the /proc and /sys "files". You could try running the sensors-detect command, and see if it adds anything to /etc/modules. If it does, then you may be in luck. Don't forget that in order for Vitals to see the newly populated files in /proc and /sys, you have to either log out/log back into your DE, reboot, or do alt-F2, r, then enter. I'm going to close this ticket to keep things tidy but if you have another question please let me know.

closed 2021-03-18 by corecoding