corecoding/Vitals #186: Gnome Shell 40 support

Tuxino88, 2021-03-19:
vitals will be available for gnome 40?

excellent extension

corecoding, 2021-03-19:
I don't see why not, but I haven't checked gnome 40 for compatibility either. Is it not working?

Tuxino88, 2021-03-19:
> I don't see why not, but I haven't checked gnome 40 for compatibility either. Is it not working?

gnome-extension-app tells me it's not compatible :S

Tuxino88, 2021-03-20:
[image]

trgeiger, 2021-03-22:
I also believe that Gnome 40 is requiring that the configuration window for extensions is written in GTK4. If I bypass the Gnome shell version check to use Vitals, the configuration/settings window doesn't work.

corecoding, 2021-03-22:
> I also believe that Gnome 40 is requiring that the configuration window for extensions is written in GTK4. If I bypass the Gnome shell version check to use Vitals, the configuration/settings window doesn't work.

This is good info. But does the main extension itself work?

I apologize for not knowing - I run Ubuntu and they aren't using gnome 40 yet, even in 21.04. I'll have to install another distro soon.

trgeiger, 2021-03-22:
It used to earlier in the Gnome 40 release cycle but it doesn't now. Gives an "ExtensionUtils.versionCheck is not a function" error. I haven't looked into it much more than that.

Tuxino88, 2021-03-22:
> It used to earlier in the Gnome 40 release cycle but it doesn't now. Gives an "ExtensionUtils.versionCheck is not a function" error. I haven't looked into it much more than that.
> 
> If you want an easy VM candidate for testing Gnome 40, you can find Fedora 34 isos now: https://dl.fedoraproject.org/pub/alt/stage/34_Beta-1.3/Workstation/x86_64/

I added the 40.0 version in metadata.json to be able to have an output. The problem behind imho is too much speed in updating, gnome like other projects. You should take your foot off the accelerator by limiting yourself to the bugfix for longer in order to give the developers breath and calm. Running after a regression every time I understand it's maddening.

Tuxino88, 2021-03-22:
> It used to earlier in the Gnome 40 release cycle but it doesn't now. Gives an "ExtensionUtils.versionCheck is not a function" error. I haven't looked into it much more than that.
>
> If you want an [...]

I added version 40.0 in metadata.json in order to have an output. The problem behind imho is too much speed in the alarm, gnome like other projects, a new DE every 6 months is not conceivable. They should take their foot off the accelerator by just fixing the bug longer to give the developers some breathing room and calm. Running after a regression every time I realize it's maddening.

kouros174, 2021-04-08:
It is the only extension I can not live without. Please upgrade to gnome-40.

ZauJulio, 2021-04-08:
I am looking forward to GTK4 / Gnome 40 compatibility! I can try to collaborate as soon as I have a spare time, but I'm sorry to inform you that it may take some time...

kouros174, 2021-04-08:
I will wait patiently. Thanks a lot! :)

abhaypatil2000, 2021-04-09:
Commenting to show support for gnome 40 port.

corecoding, 2021-04-09:
Thank you for the feedback. I will take a look this weekend. My Ubuntu partition got corrupt after using Clonezilla, so it's a good time to install a Gnome 40 distro.

xoan, 2021-04-10:
> This is good info. But does the main extension itself work?
>
> I apologize for not knowing - I run Ubuntu and they aren't using gnome 40 yet, even in 21.04. I'll have to install another distro soon.

Extension itself **works pretty well** on GNOME 40 (ExtensionUtils.versionCheck is not available anymore, so Config.PACKAGE_VERSION should be parsed manually), but prefs window seems that should be rewritten completely (gtk4, no packing, no margin, no shadow, etc...).

There is a good porting guide: https://gjs.guide/extensions/upgrading/gnome-shell-40.html

Eskander, 2021-04-10:
Can confirm that after bypassing `ExtensionUtils.versionCheck` on [L76](https://github.com/corecoding/Vitals/blob/master/extension.js#L76) and [L522](https://github.com/corecoding/Vitals/blob/master/extension.js#L522), the extension works as expected.  Aside from the `prefs` window of course.

corecoding, 2021-04-11:
Thank you @Eskander and everyone for their input. I started work tonight by installing Fedora 34. It was buggy because it was using Wayland on my Nvidia card, and the desktop kept locking up. After managing to do some work, I realized @coadmunkee created a PR but closed it with the changes in the prefs.ui. Unfortunately it looks like I have to drop support for 3.38 and below. 👎🏻  Anyhow, if any of you are brave enough to try the develop branch, I'd be much obliged.

maop, 2021-04-11:
Hello, i changed to 'develop' branch and it's working again in voidlinux+gnome40
[image]

ZauJulio, 2021-04-11:
**Confirmed**, everything working as it should!

How to make? Just follow the README tutorial of the develop branch

[image]

[image]

kouros174, 2021-04-11:
I confirm that works in manjaro too. :)

corecoding, 2021-04-11:
Thanks for confirming. New  version has been uploaded. It usually takes anywhere from a day to a week to get approved/released.

abhaypatil2000, 2021-04-11:
> Thank you @Eskander and everyone for their input. I started work tonight by installing Fedora 34. It was buggy because it was using Wayland on my Nvidia card, and the desktop kept locking up. After [...]

As far as I know, you can download the extension for the older shells right from extensions.gnome.org. There is a drop down menu to select the shell version. So not an issue.

coadmunkee, 2021-04-11:
I just submitted one more PR. I had closed the prior PR because I found a few more issues with the prefs.us and prefs.js changes that were required for GTK4 that I needed to fix. With the latest changes, everything should work as close as possible to the prior version of Vitals.

ludenticus, 2021-04-20:
Nevermind. I was missing the gnome-icon-theme-symbolic package. Great job you do.

corecoding, 2021-04-21:
@coadmunkee I believe your second PR is in master/develop. Can you confirm?

closed 2021-04-21 by corecoding

coadmunkee, 2021-04-21:
confirmed. everything looks good. my second round of changes to clean up prefs are showing in master and devel