corecoding/Vitals #187: Disable no longer available (but enabled) sensor

Mershl, 2021-04-08:
**Describe the bug**
I've configured Vitals to show Wi-Fi link quality. After disabling the Wi-Fi card the sensor value is shown as "..." in the top panel. Trying to disable the no longer needed sensor seems not possible - as it is no longer offered in the dropdown menu.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable Wi-Fi link quality.
2. Disable/Remove Wi-Fi hardware.
3. Lock and Unlock Gnome Session
4. Top Panel shows previously shown Wi-Fi link quality as "...".
5. Drop down menu does no longer list Wi-Fi link quality.
6. No (obvious) way to disable Wi-Fi link quality, now continuously shown as "...".

**Expected behavior**
Wi-Fi link quality can be disabled in the dropdown menu while it's enabled, while the sensor is no longer available.

**Screenshots**
Wi-Fi hardware available and Wi-Fi link quality sensor enabled:
[image]

Wi-Fi hardware disabled, while Wi-Fi link quality sensor is still enabled, no (obvious) way to disable it:
[image]

**Desktop information:**
 - OS: Fedora 33
 - Gnome Version 3.38

corecoding, 2021-04-21:
If you click the round refresh icon, does it go away?

Mershl, 2021-04-22:
Unfortunatly that is not the case. Clicking the circular refresh button (with the circled arrow) on the bottom left of the popup window does not make the no longer available sensor go away. The sensor stays "selected" and is continously shown as "...".

The issue can be easily reproduced on a notebook: Select Wifi Quality as a sensor for the top bar in Vitals. Reboot into BIOS and disable the internal Wifi card. Boot and see the sensor now shown as "..." with no obvious way to disable it.

corecoding, 2021-06-24:
I think I misunderstood what you were first saying. If you click another sensor/value to put it in the top bar, then click it again to hide it, it should remove the missing WiFi value. This is a short term fix (hopefully) until we can figure out a way to better handle this.

I do believe there are cases where other users would want "..." to show in case they turn their WiFi on (or whatever intermittent sensor they are using). Maybe going back to the Refresh button idea, if clicked it would prune zero value sensors (mentioned in another GitHub issue) and "..." values from the top.

corecoding, 2023-03-08:
Closing this ticket in favor of #287 - Will likely be working on this for the next release

closed 2023-03-08 by corecoding