corecoding/Vitals #191: Vitals crashes/be disabled on Gnome 40

Mershl, 2021-05-09:
**Describe the bug**
After enabling the extension it will be active for 30 seconds, after which it will be disabled by GNOME.

**To Reproduce**
- Enable Vitals via Gnome Extensions
- Wait 30 seconds
- Vitals is no longer shown. Is listed as Disabled in Gnome Extensions. 

**Log in the moment of GNOME disabling Vitals**
```
gnome-shell[1633]: Usage of object.actor is deprecated for VitalsMenuButton
                                               get@resource:///org/gnome/shell/ui/environment.js:317:29
                                               _init@/var/home/mershl/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:74:9
                                               wrapper@resource:///org/gnome/gjs/modules/script/_legacy.js:83:27
                                               enable@/var/home/mershl/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:517:18
                                               _callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:167:32
                                               _onEnabledExtensionsChanged/<@resource:///org/gnome/shell/ui/extensionSystem.js:493:35
                                               _onEnabledExtensionsChanged@resource:///org/gnome/shell/ui/extensionSystem.js:493:14
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8ad84440), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8906d440), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8ae78790), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8a1ed750), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da89624990), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8a167c40), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8b4b4500), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8a578080), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8ae50840), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8addb650), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8b6353b0), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8b4cf160), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8b149910), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8912e460), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da89414920), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da89839aa0), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8aca7e20), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da898f95e0), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8aca5d20), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da89898e50), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), o [... cut ...]
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
gnome-shell[1633]: Object .Gjs_Vitals_CoreCoding_com_menuItem_MenuItem (0x55da8b4e6840), has been already deallocated — impossible to access it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
gnome-shell[1633]: clutter_actor_destroy: assertion 'CLUTTER_IS_ACTOR (self)' failed
```

**Desktop (please complete the following information):**
 - OS: Fedora 34
 - Gnome Version 3.40.1
 - Vitals Version: 40

corecoding, 2021-05-17:
It is interesting that you said this happens only when upgrading to Fedora 34. I think what is happening is that the error is because you are actually using the previous version of Vitals that was locked at Gnome 3.38. Starting with Gnome 40, another release / version of Vitals is being used. For example, if I were to fix bugs/add features, users of 3.38 would not see them.

Does this still happen if you uninstall Vitals and re-install it?

Mershl, 2021-05-17:
Hi @corecoding, I still have both installations (one upgraded F33->F34, on clean installed F34). Both systems show the same version string `40` in `gnome-extensions info Vitals@CoreCoding.com`.

I confirmed the issue on the upgraded system again. The extension was disabled after ~30 seconds and the logs as shown in the initial post are logged.

After your suggestion I've uninstalled Vitals, made sure it's completly gone in .local/share/... (there were no remains) and re-installed it. It has been running on the upgraded system for 10 minutes without error now. Thank you very much!

closed 2021-05-17 by Mershl