corecoding/Vitals #193: "Hide zero values" does not work

lonble, 2021-06-17:
**Describe the bug**
"Hide zero values" does not work

**To Reproduce**
Steps to reproduce the behavior:
1. show network information
2. disconnect network
3. all upload/download speeds are zero
4. See error

**Expected behavior**
network information should be hidden

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [ArchLinux]
 - Gnome Version [40.2.0]

corecoding, 2021-06-18:
The hide zero values feature was not intended to hide items in the menu bar. Imagine metrics appearing and disappearing each time they go from zero to non zero. It would be quite distracting. The feature is instead intended to not show values which stay at zero forever. If at any point a sensors goes above zero, it will stay in the menu drop down and menu bar.

lonble, 2021-06-19:
When the network is disconnected, the network speed will always be zero.
I expect that when I connect to the wifi, the network speed will be displayed; when I disconnect the wifi, it will be hidden.

corecoding, 2021-06-21:
I appreciate your input. Unfortunately, that is not how I designed the hide zero values feature. If I were to implement this change, I would have to know whether a network device was disconnected, or simply at idle. I am not aware of a way using /proc or /sys directories to know whether a device is connected or not. 

Also, don't forget that this feature is not just for networking. It's for fan, temperature, and every other sensor. I'd have to come up with logic to handle each of those differently.

I have been thinking of a related feature, which is when you click refresh (round arrow circle) it removes values that are currently at zero. I think this would help a lot of other users of Vitals, and maybe your case too.

closed 2021-06-21 by lonble