corecoding/Vitals #196: Pop! OS Cosmic error

ScizorsBLBC, 2021-07-14:
The newest update to the extension is not working on Pop! OS newest update of cosmic. 
[image]

corecoding, 2021-07-14:
Thank you for reaching out. I installed Cosmic after seeing your comment and Vitals is working for me. 

Can you take a look at /var/log/syslog and search for Vitals? If I can get a stack trace I can likely fix the issue.

ScizorsBLBC, 2021-07-14:
Hello, I think this is what you were asking for, I am not sure tho. I am
not the most experienced linux user.

Jul 13 19:04:01 pop-os dbus-daemon[1799]: [session uid=1000 pid=1799]
Activating service name='org.gnome.ChromeGnomeShell' requested by ':1.439'
(uid=1000 pid=29908 comm="/usr/bin/python3 /usr/bin/chrome-gnome-shell
chrom" label="unconfined")
Jul 13 19:04:01 pop-os dbus-daemon[1799]: [session uid=1000 pid=1799]
Successfully activated service 'org.gnome.ChromeGnomeShell'
Jul 13 19:04:31 pop-os gnome-shell[30026]: Archive:
 /tmp/46N450.shell-extension.zip
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/extension.js
Jul 13 19:04:31 pop-os gnome-shell[30026]:    creating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/helpers/
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/helpers/file.js
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/helpers/polyfills.js
Jul 13 19:04:31 pop-os gnome-shell[30026]:    creating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/battery-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/cpu-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/fan-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/memory-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/network-download-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/network-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/network-upload-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/storage-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/system-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/temperature-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/icons/voltage-symbolic.svg
Jul 13 19:04:31 pop-os gnome-shell[30026]:   inflating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/LICENSE
Jul 13 19:04:31 pop-os gnome-shell[30026]:    creating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/locale/
Jul 13 19:04:31 pop-os gnome-shell[30026]:    creating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
/locale/ca/
Jul 13 19:04:31 pop-os gnome-shell[30026]:    creating:
/home/scizors/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com
[... 170 lines cut ...]
Vitals@CoreCoding.com: Error: A different version was loaded previously.
You need to log out for changes to take effect.#012loadExtension
@resource:///org/gnome/shell/ui/extensionSystem.js:343:52#012callback
@***@***.***:///org/gnome/shell/ui/extensionDownloader.js:115:13
Jul 13 19:05:05 pop-os /usr/libexec/gdm-x-session[2222]: (EE) event4  -
Logitech MX Master 3: client bug: event processing lagging behind by 11ms,
your system is too slow
Jul 13 19:05:05 pop-os /usr/libexec/gdm-x-session[2222]: (EE) event4  -
Logitech MX Master 3: WARNING: log rate limit exceeded (5 msgs per 60min).
Discarding future messages.
Jul 13 19:05:06 pop-os touchegg[979]: libinput error: event4  - Logitech MX
Master 3: client bug: event processing lagging behind by 11ms, your system
is too slow
Jul 13 19:05:07 pop-os touchegg[979]: libinput error: client bug: timer
event4 debounce: scheduled expiry is in the past (-0ms), your system is too
slow
Jul 13 19:05:07 pop-os touchegg[979]: libinput error: client bug: timer
event4 debounce short: scheduled expiry is in the past (-13ms), your system
is too slow
Jul 13 19:05:07 pop-os /usr/libexec/gdm-x-session[2222]: (EE) client bug:
timer event4 debounce: scheduled expiry is in the past (-0ms), your system
is too slow
Jul 13 19:05:07 pop-os /usr/libexec/gdm-x-session[2222]: (EE) client bug:
timer event4 debounce short: scheduled expiry is in the past (-13ms), your
system is too slow

corecoding, 2021-07-14:
Yes, this is exactly what I was looking for. Thank you. The specific error you are receiving is:
"A different version was loaded previously. You need to log out for changes to take effect."

I've had a few reports of this in the past when I updated Vitals. If you log out of your desktop and log back in, Vitals should start working. Worst case, if you uninstall it by clicking the red X, and re-install it as you did initially, it should start working again. It should even keep your previously selected menu bar sensors.

I'll ask the Gnome guys about this error so I can try to prevent it in the future.

ScizorsBLBC, 2021-07-14:
Hey that worked, I did a reboot and now it's working as it did before, even has the previously selected preferences as you suspected it would!  Thank you for resolving this so quickly!

closed 2021-07-14 by ScizorsBLBC