corecoding/Vitals #199: Battery/Voltage showing No Data though GNOME Power Statistics fully populated

Saijin-Naib, 2021-08-20:
**Describe the bug**
I read the wiki entry about how Vitals polls the /proc/sys/ directory. However, GNOME Power Statistics fully populates power information for my platform, though I'm not sure what/how it reads this.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Vitals
2. Click on Voltage or Battery
3. Observe No Data
4. Open GNOME Power Statistics
5. Observe full data

**Expected behavior**
Vitals will be able to report battery/voltage information that my platform is reporting to GNOME.

**Screenshots**
[image]

**Desktop (please complete the following information):**
Alpine Linux Edge 3.14.1 with Linux-LTS
GNOME 40.4

corecoding, 2021-08-20:
If you go into Vitals settings, you can change the battery slot that it references. There is a cog icon near battery. Please let me know if you tried this.

Saijin-Naib, 2021-08-20:
> If you go into Vitals settings, you can change the battery slot that it references. There is a cog icon near battery. Please let me know if you tried this. 

Yeah, I tried all the listed battery slots BAT0, BAT1, CMB0, etc. No change.

Saijin-Naib, 2021-08-20:
Yep, I tried BAT0 through CMB0. No change.

corecoding, 2021-09-20:
Can you share some of your directory and sub directories under /sys/class/power_supply/ ?

corecoding, 2021-10-04:
If you take a look at the link below, you can see the alternate path when the file/sensor "charge_full" doesn't exist. The alternate path checks files energy_full, voltage_min_design, energy_full_design, voltage_now, power_now, and energy_now. According to your screenshot, you do have those files so it should be working on BAT0.

https://github.com/corecoding/Vitals/blob/c2656f895ab5f6c7e27cdc49149ea54bee23b393/sensors.js#L304-L331

It is really hard to debug an issue like this when I don't have access to your hardware. If you want to attempt to debug it yourself, the first thing you have to do is use global.log('testing'); for example to step through the code to see if it is even getting to that section. With Gnome, it is a little hard to debug especially when using Wayland, as each change to sensors.js you have to restart Gnome Shell to see those changes. The output of global.log goes to /var/log/messages (or syslog)

stark1tty, 2022-02-22:
I am having the same issue.

corecoding, 2022-12-11:
I have changed how the battery status is checked in the app. This was part of issue #301. The change wasn't intended to address this ticket, but there is a chance it might. If this doesn't fix this ticket, as I said above I will need someone to create a PR with code changes, as I don't have access to your hardware/system. Going to close this ticket, hoping that the issue above fixes it - otherwise I'll open this ticket back up when I see a PR.

closed 2022-12-11 by corecoding