corecoding/Vitals #200: How to add custom sensors on gnome top bar

prateekmedia, 2021-08-24:
**Describe the new feature you'd like**
I want to add netspeed icon on top panel with vitals and add some other things like cpu, temperature etc, by my own but I can't find a option to do that in vitals.

**Additional context**
Most of the system monitors have it, I have used many on other desktop environments like pantheon, budgie, etc. It would be nice addition to vitals to add them by own and remove the ones which I don't want.

It would be to add them in settings/ preferences of vitals like a button or checkbox saying add to top panel, and maybe a way to arrange them.

corecoding, 2021-08-24:
This should help you out: https://github.com/corecoding/Vitals/wiki/Can't-add-sensors-to-panel

closed 2021-08-24 by corecoding

prateekmedia, 2021-08-24:
@corecoding This should be on the readme and extensions gnome page, as no one would ever notice that and its not intuitive to the user.

corecoding, 2021-08-24:
Appreciate the feedback. Adding to the extensions page and read me won't help as much as you think. Most people don't go to GitHub. Anyhow, there is a ticket over at https://github.com/corecoding/Vitals/issues/176 where others have asked for the same thing. Ideally an overlay / instruction when the extension is installed would be best.

Vilasamuni, 2021-11-21:
I was about to open a feature request when I saw this. I looked in the wiki but I didn't notice the navigation on the right - only the links in the main body of the wiki page.