corecoding/Vitals #209: Icon for System Monitor missing

J3RN, 2021-11-10:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

The icon for the link to launch "System Monitor" is missing 
[image]

### Steps to reproduce the unexpected behavior.

1. Enable extension.
2. Click indicator in the notification tray.
3. Icon for the middle circle in the bottom ("System Monitor") is missing.

### What distribution and version of Linux are you using?

Fedora 35 Workstation

### What version of Gnome are you using?

GNOME 41.1

asarro99, 2021-11-12:
Hi, I just signed up for a pull request to fix the problem #210.
If you want to solve it as soon as possible, you just need to install an alternative icon pack like Tela that has the missing icon inside.
I hope I have helped you

corecoding, 2021-11-13:
Thank you everyone! I have merged this to the develop branch, and will move this ticket into the next Project (Release)

J3RN, 2021-11-13:
Tremendous! Thank you both! :clap:

closed 2021-11-13 by J3RN