corecoding/Vitals #212: Display freezing for a split second when reloading

SIMULATAN, 2021-11-30:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

I just did a clean install of manjaro with gnome on my laptop and installed the vitals extension but unfortunately, contrary to what the description says it isn't async for me. Every 5 seconds (which is the reload delay), the whole display freezes and after like 200 milliseconds the vitals are updated and everything else continues.

### Steps to reproduce the unexpected behavior.

1. Download the Plugin

### What distribution and version of Linux are you using?

Manjaro 21.1.6

### What version of Gnome are you using?

41.1

corecoding, 2021-12-01:
This is obviously not what is expected from Vitals. I think this issue https://github.com/corecoding/Vitals/issues/150#issuecomment-645016570 may be related. I think what is happening is that there is a sensor in /sys (possibly /proc) that when accessed, causes the system to pause for 200ms. Does the issue go away if you turn off one of the sections (network, temperature, etc) ? It may be a challenge to figure out exactly what sensor is causing it.

SIMULATAN, 2021-12-01:
Thanks for the quick answer!
I turned off every sensor one by one in the extension settings but it keeps updating, I think I'm doing something wrong...

closed 2021-12-01 by SIMULATAN