corecoding/Vitals #214: Gpu stats

gfrcr, 2021-12-08:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

I want to know if there is any config that I could make to show the gpu stats on vitals

corecoding, 2021-12-10:
https://github.com/corecoding/Vitals/wiki/Can-you-add-support-for-my-GPU%3F

corecoding, 2021-12-10:
There is no config that can bridge the gap from a command line utility to the vitals UI. Of course you can modify the source code to run the command - but as the Wiki says, it's not something that the public version of Vitals will do.

closed 2021-12-10 by corecoding