corecoding/Vitals #216: Add the sum of rates of all network interfaces

lonble, 2021-12-10:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Currently there is a maximum value in the network label, but this value is not often used.  What I need to know is the rate of each interface or the sum of the rates of all interfaces.  Please add this statistic.

corecoding, 2022-02-18:
So you are saying that if your WiFi interface was running 10MB Down and 1MB up, you want a label called WiFi Sum with a value of 11MB?

lonble, 2022-02-18:
> So you are saying that if your WiFi interface was running 10MB Down and 1MB up, you want a label called WiFi Sum with a value of 11MB?

No. I have both ethernet and wifi connected, ethernet uploads 10MB/s and wifi uploads 1MB/s, I need a total upload value of 11MB. This is also required for download rate.

corecoding, 2022-02-19:
Will be included in the next update.

closed 2022-02-19 by corecoding