corecoding/Vitals #217: Add the option of whether to include lo in the network statistics

lonble, 2021-12-10:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Before entering the topic, the upload and download of lo are always the same, so there is no need to divide it into two sensors.

Next is the topic. 
In many practical scenarios, data packets need to be sent to the processing program of the local machine through lo, and then sent to the outside, such as the local proxy. At this time, I cannot get valuable information from the Maximum, because it is covered by lo data, and upload and download are always the same. 
I also mentioned the sum in Network in my last issue. It would be helpful if lo can be removed from these two statistics.
Of course, some people may have different usage scenarios and expect lo to be included in the statistics. I suggest adding an option to control this feature.

corecoding, 2022-01-29:
So just to confirm, you are asking for an option to exclude lo from Maximum?

lonble, 2022-01-29:
Exactly.

corecoding, 2022-01-31:
I pushed changes to the develop branch that simply ignore 'lo'. I was going to add a checkbox to include or exclude lo, but I think most people wouldn't know what that is, or, never turn it on.

I will likely wait for Gnome 42 before releasing this feature - I'm assuming I may have to tweak Vitals to add support.