corecoding/Vitals #220: Add capability to open a system monitor other than gnome-system-monitor

sp-merrow, 2021-12-25:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Hi, I'm requesting a feature be added where when you click the middle button at the bottom of the Vitals dropdown menu (the button that opens the system monitor), you can configure it to open a system monitor other than gnome-system-monitor. Something like a field in the options menu where you can type in the command to run to open the system monitor. I do not have gnome-system-monitor installed because I don't use it (I use btop as my system monitor), so currently clicking on the button that normally opens gnome-system-monitor does nothing.

Thanks for your consideration, and thank you for the work you do on this fantastic extension!

Jose-Verdu-Diaz, 2022-06-14:
Any update on this issue? I would also love to see this feature implemented.

luixal, 2022-07-28:
I've implemented this. Can you take the code from the referenced PR and test it?

Thanks!

corecoding, 2022-08-16:
This feature was changed slightly - the configuration was moved into the "Monitor System" section. But other than that, the functionality is the same. It was released today!

closed 2022-08-16 by corecoding