corecoding/Vitals #221: CPU speed units and decimal places

spxak1, 2021-12-27:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Setting the units to GHz or MHz has been asked in #181 

This feature request also adds the decimal places. While one decimal place is needed when the speed is reported in GHz, it is certainly redundant when in MHz.

The two feature requests can be combined. 

The reason for the request is that even with fixed width, there is a constant jump every time the units change from GHz to MHz, made even larger by the extra digits of the MHz readings.

Effectively it's all about that constant change in the display on the panel. Thanks again.

canadaduane, 2022-05-04:
Similarly, the network speed (MB/s, KB/s) suffers from this issue.

corecoding, 2022-05-05:
I marked #181 as a duplicate of #168