corecoding/Vitals #222: Hide unwanted sensors

sql-sith, 2021-12-29:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

This could be merged with #83 conceptually, under the idea of "label customization." In addition to being able to change the text on a label (that's #83), I would also like to be able to shut off sensors I just don't care about. I mean, it's great to know that TCMc reports that it's running at -128 Centigrade, but ... wait, no, it's not great.

Seriously, I love the interface, but there's just too much info. And no matter how little I detect with `sensors-detect`, all of these sensors always get detected. I'd really like to hide them. Thanks for considering!

[image]

Haptein, 2022-02-20:
If I may suggest, rather than hiding sensors, not reading from them in the first place might be better. But I agree, from tabs like processor, temperature and network I'm interested in at most a couple of readings and the rest can feel like bloat.

HMPoweredMan, 2022-04-15:
I just wanted to add to this. asus_wmi temp sensors report 216 for sensors not present on a motherboard. Because of this the ability to exclude them would be great at they throw off the average significantly.

[image]

espensa, 2025-06-09:
I would love to see this implemented, I have some sensors throwing off the average, I tried hiding them from lm_sensors but your extension still showed them

code-mha, 2026-10-07:
This is much needed for tidying up system monitoring. Please make it possible. Thanks.