corecoding/Vitals #224: The dropdown for Vitals isn't centered

thatLeaflet, 2022-01-15:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

When clicking on the vitals menu, it shows its dropdown off to the right of the widget. The normal behavior of the gnome widgets, such as the calendar/time widget, would be to have the dropdown centered below the widget.

Image of current dropdown: https://imgur.com/a/sxb9sh2

### Steps to reproduce the unexpected behavior.

1. Click on widget

### What distribution and version of Linux are you using?

Fedora 35

### What version of Gnome are you using?

Gnome 41.3

corecoding, 2022-02-21:
The dropdown fits the screen automatically, which is handled by Gnome itself. If Vitals were on the far right (other side of Gnome's own dropdown, the arrow would be on the other side.

closed 2022-02-21 by corecoding

corecoding, 2022-02-23:
I stand corrected, you can change the arrow position. I pushed some changes that makes it adjust to the position where vitals is at. So for example, center, the arrow is center. Left, the arrow is right, and right, the arrow is left. Maybe not quite what you wanted but I love simplicity. :)