corecoding/Vitals #225: this._sensorIcons[type] is undefined

emilydolson, 2022-01-15:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Vitals does not show up, and cannot be enabled from the Extensions menu. Instead, there is a red exclamation point next to it and it displays the error listed below. I messed around with debugging it myself a bit and was able to re-enable it after forcing `type` to always be "processor", but I don't know much about writing GNOME extensions and couldn't get much farther.

### Steps to reproduce the unexpected behavior.

1. Install vitals
2. Go to Extensions
3. See error and inability to enable Vitals

### Relevant log output

```shell
this._sensorIcons[type] is undefined
```

### What distribution and version of Linux are you using?

Pop!_OS 21.10

### What version of Gnome are you using?

40.4.0

corecoding, 2022-01-29:
It is hard to know exactly what is going on because I don't have a stack trace or line number. type has to be unexpected, empty, etc around the line number that it is occurring on. If you could add global.log(type) on the line before the error, and tell me the value you see in your logs, it would be helpful!

corecoding, 2022-02-13:
We can reopen once additional data is attached.

aderbane, 2022-02-23:
Hi Chris,

First of all, thank you for this extension! Up until  I started to have the exact same issue as described above, I used to love Vitals. I want to try to help you resolve the issue. This will be my first comment on an issue (ever) so I do apologise in advance if any vital information is missing. If so, please let me know and I will happily provide additional information.

### What distribution and version of Linux are you using?
Manjaro Linux 21.2.3 (5.16.7-1-MANJARO)

### What version of Gnome are you using?
41.3

### Steps to reproduce the unexpected behavior.
Identical to what was described by Emily

### Relevant log output
As asked, I went ahead and added the global.log(type) to the line before the error. Below seem to be the useful log lines:

```
feb 23 08:46:08 aliderbane-pc gnome-shell[1782]: memory
feb 23 08:46:08 aliderbane-pc gnome-shell[1782]: temperature
feb 23 08:46:08 aliderbane-pc gnome-shell[1782]: processor
feb 23 08:46:08 aliderbane-pc gnome-shell[1782]: max
feb 23 08:46:08 aliderbane-pc gnome-shell[1782]: JS ERROR: Extension Vitals@CoreCoding.com: TypeError: this._sensorIcons[type] is undefined
                                                 _defaultIcon@/home/aliderbane/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:456:13
                                                 _createHotItem@/home/aliderbane/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:229:25
                                                 _drawMenu@/home/aliderbane/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:320:18
                                                 _init@/home/aliderbane/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:60:14
                                                 enable@/home/aliderbane/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:517:18
                                                 _callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:166:32
                                                 loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:349:26
                                                 _loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:587:18
                                                 collectFromDatadirs@resource:///org/gnome/shell/misc/fileUtils.js:27:28
                                                 _loadExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:566:19
                                                 _enableAllExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:596:18
                                                 _sessionUpdated@resource:///org/gnome/shell/ui/extensionSystem.js:627:18
                                                 init@resource:///org/gnome/shell/ui/extensionSystem.js:55:14
                                                 _initializeUI@resource:///org/gnome/shell/ui/main.js:265:22
                                                 start@resource:///org/gnome/shell/ui/main.js:162:5
                                                 @resource:///org/gnome/shell/ui/init.js:6:17
```
In case you might need it I will also paste the function in which I have placed the global.log so you have some more context:

```js
    _defaultIcon(key) {
        let split = key.replaceAll('_', ' ').trim().split(' ')[0].split('-');
        let type = split[0];

        let icon = new St.Icon({
          style_class: 'system-status-icon vitals-panel-icon-' + type,
            reactive: true
        });

        // support for hide icons #80
        if (type == 'default') {
            icon.gicon = Gio.icon_new_for_string(Me.path + '/icons/' + this._sensorIcons['system']['icon']);
        } else if (!this._settings.get_boolean('hide-icons')) {
            let iconObj = (typeof split[1] != 'undefined')?'icon-' + split[1]:'icon';
            global.log(type)
            icon.gicon = Gio.icon_new_for_string(Me.path + '/icons/' + this._sensorIcons[type][iconObj]);
        }

        return icon;
    }
```

Feel free to let me know if more information is needed. 

I also decided to give you a small donation through Paypal. Even if you were to never solve this issue, it will be a donation for the times I enjoyed with Vitals, before it stopped working for me. 

Thank you kindly again for all the hard work you've done and do! :)

corecoding, 2022-02-23:
I think I found the issue. Way back in May 6, 2020, [I made this change](https://github.com/corecoding/Vitals/commit/eb8ced482d16ab8e8ac940ad42d25f7d43516049). 

As you can see, I moved _max_ to the end. Will keep you posted.

corecoding, 2022-02-24:
As far as the code above, you are right. I actually edited my comment to:
```
        if (type == 'default' || !(this._sensorIcons[type])) {
```
But your `global.log(key);` was all I needed to fix the issue. I submitted a new version of Vitals to the Gnome Extensions portal tonight. Hoping it'll be available for you to officially test tomorrow. Please let me know if it works for you.

As far as the other issue, the value next to Network is your Total (maybe combined is a better name) Download traffic. So if you had two interfaces both downloading at 5mb/s, it would show 10mb/sec. To test it out, try going to speedtest.net and compare the speeds. If you don't want to do calculation between bytes and bits, you can set vitals to report bits instead. If you are seeing different speeds next to Network when downloading, I think that would be a separate bug report/issue.

corecoding, 2022-02-24:
Also, forgot to mention that the Network section will never be identical to speedtest.net. Reasons for this is the update interval may include time before or after the speed test, bloat on communication headers, and other network processes besides your browser. But again it should be close. Vitals pulls the numbers from the kernel, it really doesn't do any calculations other than for bits/bytes.

aderbane, 2022-02-24:
Chris, I have just installed the shiny updated version of Vitals. And for some reason it feels like such a relief that it works as it used to. Everything seems to work as expected. From my point of view I suppose this issue can be closed!

Regarding the other "issue" I pointed out, I will add two screenshots. The same one I added in my previous comment but with red (the "issue") and green (probably the part you commented about)  markings showing what I meant. I feel I was not very descriptive in the "issue" I had. 
Having said that though, the second screenshot will show you that the fix you uploaded, also fixed that issue.

[image]
_before_

[image]
_after_

Thank you so much for your hard work and swift and friendly communication. Keep up the good work! :)

corecoding, 2022-02-24:
Glad to hear! Thank you so much for your help. On eBay you would receive "A+++ would do business again!" response.

closed 2022-02-24 by corecoding

aderbane, 2022-02-24:
Haha, you are very welcome! In gamers world you would hear me say "gg, wp" :)