corecoding/Vitals #228: Fix CPU crequency to GHz

im-AMS, 2022-02-04:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Currently the CPU frequency uses both MHz and GHz. when its in MHz(like 900Mhz) its a 3 digit and when in GHz (like 2.5GHz) its 2 digit, this causes the widget to keep moving around always and it gets annoying.

**Suggested**
- Either set a constant digit aka fix the width to constant so that it does not keep moving
- Fix the CPU frequency to always show in GHz (This seems more elegant to me)

corecoding, 2022-02-14:
There is actually a "Use fixed widths" checkbox in Preferences that will stop it from moving. 

As far as the second suggested option, I kind of like this idea. I'm going to leave this ticket open for a bit and see if anyone else chimes in.

corecoding, 2022-02-17:
I am closing this ticket in favor of #181. Please see communication over there.

closed 2022-02-17 by corecoding