corecoding/Vitals #234: Network - total downloaded/uploaded

den7007, 2022-02-13:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

#229 

Wondering if possible to include total amount downloaded/uploaded for the session as another item under Network? And maybe an option to reset the counter whenever we choose?

corecoding, 2022-02-19:
Sorry, I know I asked you to create this second request, but I realized it is a duplicate of #216. Good news is that's the feature I'm working on today!

den7007, 2022-02-21:
> Sorry, I know I asked you to create this second request, but I realized it is a duplicate of #216. Good news is that's the feature I'm working on today!

Hi, was looking for total amount downloaded/uploaded for the session, not total download/upload speed, if that makes sense.

Something similar to what is implemented here (5th mode): https://extensions.gnome.org/extension/3724/net-speed-simplified/

corecoding, 2022-03-09:
This has been added to the latest master branch. It should be released within a week.

closed 2022-03-09 by corecoding