corecoding/Vitals #236: "Hide zero values" is enabled but a non-existent fan with 0 rpm is displayed and contributes to average

lpinner, 2022-02-15:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

"Hide zero values" is enabled but a non-existent fan with 0 RPM  is displayed and contributes to average.

This motherboard has never had a 2nd case fan plugged in, just the cpu cooler fan and a single case fan.

[image]

So instead of an average RPM of (1418+1383)/2 = 1401, the average is (1418+1383+0)/3 = 937

### Steps to reproduce the unexpected behavior.

1. Enable "Hide zero values".
2. Either unplug a fan, or do not install one in the first place.

### What distribution and version of Linux are you using?

Ubuntu 21.10

### What version of Gnome are you using?

GNOME Shell 40.5

corecoding, 2022-06-16:
Today, this is expected behavior. It is because a lot of computers completely turn off fans when the computer is running cool enough. Vitals would need logic to hide the fan (if computer booted at 0 rpm) and show it the first time it turns on. I do believe once that happens, it should show the fan even at 0.

Another solution (which may be easier) is to use #222 and let the user hide the fans.

corecoding, 2022-07-10:
Will be using #222 as solution for this request.

closed 2022-07-10 by corecoding