corecoding/Vitals #237: Add a setting to hide "..." button

dmytro-shchurov, 2022-02-21:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

This is sad that #176 issue (I guess this is it) about a hint to add new censor  was solved by sacrificing of so needed space. 
[image]
For me this is too much. Few people who had no idea to click on the panel made the rest life harder.
Tray icons appear collapsed almost all the time now.
[image]

It would be nice to have "show ... button" settings option.

corecoding, 2022-02-22:
Regarding #176, it has nothing to do with space. I feel that people understand how Vitals works now with the recently added GIF in the Gnome extensions website. Also, I'd like to spend my time on some of the other more commonly requested features. Not saying this isn't a worthy one though!

Regarding this feature, you are asking for a "..." menu item that opens up the settings (preferences) page?

dmytro-shchurov, 2022-02-22:
> you are asking for a "..." menu item that opens up the settings

I'm not. I'm talking about Gnome 40 and a situation when Vitals has the `...` button, which occupies some space on the top bar, and so Tray icons extension appears always collapsed because of this. For me it would be nice to have a *settings* that hides that `...` button.

A week ago (or so) Vitals did not have that button.

PS The `...` button is placed between Vitals and locale indicator, has huge left/right margins, and it's even hard to say which extension has it.

corecoding, 2022-02-22:
The only `...` that I can think of is when you have a sensor selected to be in the menu bar that no longer exists. If you select another sensor from the drop-down menu, the `...` should be removed. That being said, I took a look at your screenshot and there is no icon next to the `...` so maybe this is not it. But do try and let me know.

If the `...` doesn't go away, I have to ask you, what happens when you click it? Does the Vitals drop-down menu appear?

dmytro-shchurov, 2022-02-22:
That would make sense, but I have only three enabled and three displayed, and the `...` button.

[image]

Thus may be this is a bug?

corecoding, 2022-02-22:
When I said click another sensor from the dropdown, I meant the Vitals menu itself. Not the preferences pane. For example, go to Processor, and click Average on then off. Do the `...` go away?

dmytro-shchurov, 2022-02-22:
I see. Now the `...` has gone. I don't think this feature request makes some sense. If there is some UI issue with that clicking/unclicking, perhaps another issue exists somewhere.

Thank you!

closed 2022-02-22 by dmytro-shchurov