corecoding/Vitals #240: Equally named sensors are merged

maxaalderink, 2022-03-01:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Since the latest update (version 51) I have noticed that I can only find one out of two NVME SSD in the temperature dropdown.
Initially I thought it only showed the drive I'm currently using as the second one is for a secondary OS.
But since lm-sensors en Psensor is still showing both I think the problem is related to Vitals.
Additionally I've seen the the temperature jump 20+C on occasions, if I had to guess that is because it sometimes reads and displays the sensor from the secondary NVME which is much cooler because it's not being used actively.

The drives uses are a Samsung 980 Pro and a Samsung 970 Evo+, for both lm-sensors detects 3 senors named Sensor 1, Sensor 2 and Composite.

### Steps to reproduce the unexpected behavior.

1. Install 2 devices that present equally named sensors to lm-sensors.
2. Run `sensors-detect`.
3. Run `sensors` to see if both devices are detected (in my case nvme-pci-0400 & nvme-pci-0100) with equally named sensors.
4. Notice that only one set of sensors can be found in Vitals.

### What distribution and version of Linux are you using?

Pop!_OS 21.10

### What version of Gnome are you using?

40.4.0

corecoding, 2022-03-07:
Acknowledged, working on it

corecoding, 2022-03-10:
Can you do me a huge favor and test out the master branch by following the beta testing instructions [over here](https://github.com/corecoding/Vitals)? Of course, on step 3 when ti says `git checkout develop` you would type `git checkout master`

maxaalderink, 2022-03-10:
> Can you do me a huge favor and test out the master branch by following the beta testing instructions [over here](https://github.com/corecoding/Vitals)? Of course, on step 3 when ti says `git checkout develop` you would type `git checkout master`

Ill try to take a look at it either tonight or tomorrow

maxaalderink, 2022-03-12:
@corecoding The main issue is resolved.
But I should note that for some reason the sensors `Composite` are now showing up as just `NVMe` while `Sensors x` are showing up as `nvme Sensor x`. 
Not a deal breaker but it does seem a bit inconsistent with the naming and to a lesser extend capitalization.

Vitals (master branch):
[image]

LM-Sensors output:
[image]

corecoding, 2022-03-17:
I was working on a feature to make the sensor names human readable. I have backed out that feature for now. The fixed release is now live. I believe your manual git clone should be overwritten automatically with the newer version.

closed 2022-03-17 by corecoding

maxaalderink, 2022-03-17:
I already installed the latest version through the conventional methods and it all seems to work fine now, including naming.

Thanks for all your work, I really like the extension :)