corecoding/Vitals #241: Please make a proper GH workflow for tagging & remove 'refs/heads/master' tag

FiestaLake, 2022-03-02:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Your current implementation made a tag 'refs/heads/master'. And it breaks the aur package called 'gnome-shell-extension-vitals-git' as it uses `git describe --tags | sed 's/-/+/g'` for versions and pkgver is not allowed to contain colons, forward slashes, hyphens or whitespace.

And it seems like your workflow is broken cause it makes a new tag with the same name which is not acceptable (without deleting the existing one)

Therefore, I recommend you to remove 'refs/heads/master' tag and make a proper GH workflow for tagging.

corecoding, 2022-03-03:
Thanks for the heads up. I have made changes to my build pipeline. I have a question / request for you.

I don't know who built the aur package. I am also not an arch user, but I am assuming there is a script of some sort that downloads the contents from my repo's master branch and builds a package for your system. Please note that when I prepare a package, I run the commands in this file: https://github.com/corecoding/Vitals/blob/f444bd23a789ae0182b9c9fe0cd534996429dd98/zip.sh

Without these commands, the preferences pane may not have the latest database schema, and you won't have the latest language changes.

FiestaLake, 2022-03-03:
> Thanks for the heads up. I have made changes to my build pipeline. I have a question / request for you.
>
> I don't know who built the aur package. I am also not an arch user, but I am assuming there [...]

You can check the commands in https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=gnome-shell-extension-vitals-git

Anyway, thanks for your quick fix. Now it doesn't give me errors now.

closed 2022-03-03 by FiestaLake