corecoding/Vitals #242: Icon changed/incorrect in top bar after update

Dorro101, 2022-03-16:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

The top bar has a temp icon with three dots adjacent, this was not the case before the update...
[image]

### Steps to reproduce the unexpected behavior.

Is a permanent graphical change, not a crash per se...

### What distribution and version of Linux are you using?

Fedora 35

### What version of Gnome are you using?

41

Dorro101, 2022-03-16:
Love your extension by the way, one of the first things I install after OS upgrade/change.

corecoding, 2022-03-16:
Did you see https://github.com/corecoding/Vitals/wiki/Sensor-shows-%22...%22-in-menu-bar ? There was a bit of a restructure in how Temp, Fan and Volt are handled.

Dorro101, 2022-03-17:
Thanks so much! Fixed!!! Now I feel like a right idiot who doesn't bother to RTFM....

Ciao
Dorian

closed 2022-03-17 by Dorro101

corecoding, 2022-03-17:
To be honest, you are certainly not the first person to ask this question. I am thinking I need to figure out a way to automatically remove "..." sensors. Of course, some may ask why the sensors disappeared, but I think it is one step in the right direction.