corecoding/Vitals #243: task bar text gone

gargozorini, 2022-03-21:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

After ticking/unticking some sensors the text in the task bar dissappeared
Neither refreshing or relogging makes the text come back
[image]

### Steps to reproduce the unexpected behavior.

Sorry, i'm noob, old and my english is less than basic

### What distribution and version of Linux are you using?

Zorin OS 16.1 Core

### What version of Gnome are you using?

3.38

blastician, 2022-03-29:
Having the same issue as well on Manjaro running Gnome 41.4. System is fully patched.

It seems like the update messed with the settings that I had in there. I was displaying CPU/ Mem usage % and that works fine but my CPU package temp. just has an icon with three dots. The other two I'm not sure what they were because they have no icon but they also have three dots. 

While troubleshooting I tried disabling and re-enabling everything and found that the ones that are not showing data are listed as not enabled. When I enabled the sensor it worked as expected but the old broken one was still there.

corecoding, 2022-03-29:
> Having the same issue as well on Manjaro running Gnome 41.4. System is fully patched.
>
> It seems like the update messed with the settings that I had in there. I was displaying CPU/ Mem usage % and [...]

The issue you are seeing is different Please take a look at https://github.com/corecoding/Vitals/wiki/Sensor-shows-%22...%22-in-menu-bar

corecoding, 2022-03-29:
@gargozorini I have tried reproducing the issue in Zorrin OS and was unable to reproduce it. Is this a stock install or did you add/remove some pages, make tweaks to the OS, etc?

gargozorini, 2022-03-29:
I installed gnome tweaks. After the issue with Vitals I disabled it and installed indicator-sensors, wich is working properly at the moment
[image]
 Enabling Vitals brings back the invisible task button, so there must be something in my configs messing with it. Thank you for answering

corecoding, 2022-04-05:
It's not going to be in the config. It's either a bug in Vitals itself, or a missing icon on your system. When I performed a fresh install of Zorrin OS, I can't duplicate the issue. Hopefully someone else can chime in.

sergiotca, 2022-04-13:
> The issue you are seeing is different Please take a look at https://github.com/corecoding/Vitals/wiki/Sensor-shows-%22...%22-in-menu-bar

After updating to V53, I thought I was dealing with the issue you linked here, so I toggled on/off all the sensors and even the extension itself but still both the memory and the system icons just show 2 dots (...):
[image]

Oddly enough, they both show the correct info on the drop down menu:
[image]

BTW, I believe the network icon on the top bar was introduce in this new release, and it is working just fine. 
Before this last update, the only two icons shown on the top bar were memory and system, and they use to work as expected.

I'm using Manjaro + Gnome 41.3 + X11

corecoding, 2022-04-14:
> > > Having the same issue as well on Manjaro running Gnome 41.4. System is fully patched.
> > > It seems like the update messed with the settings that I had in there. I was displaying CPU/ Mem usage % and that works fine but my CPU package temp. just has an icon with three dots. The other two I'm not sure what they were because they have no icon but they also have three dots.
> > > While troubleshooting I tried disabling and re-enabling everything and found that the ones that are not showing data are listed as not enabled. When I enabled the sensor it worked as expected but the old broken one was still there.
> > 
> > 
> > The issue you are seeing is different Please take a look at https://github.com/corecoding/Vitals/wiki/Sensor-shows-%22...%22-in-menu-bar
> 
> After updating to V53, I thought I was dealing with the issue you linked here, so I toggled on/off all the sensors and even the extension itself but still both the memory and the system icons just show 3 dots (...): [image]
> 
> Oddly enough, they both show the correct info on the drop down menu: [image]
> 
> BTW, I believe the network icon on the top bar was introduced in this new release, and it is working just fine. Before this last update, the only two icons shown on the top bar were memory and system, and they used to work as expected.
> 
> I'm using Manjaro + Gnome 41.3 + X11

This is a separate issue, take a look at #245 - effectively the naming in hot-sensors was changed to values, not keys. Simply toggling a sensor will remove the ... entries and you can add them back again. Going forward it should work properly. If you want to reply please do so on #245.