corecoding/Vitals #244: Sensor ... problem has reappeared

heynnema, 2022-03-25:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Temp icons showing ... in top panel. Have reset Vitals, as per previous closed report.

### Steps to reproduce the unexpected behavior.

Observe top panel icons

### What distribution and version of Linux are you using?

Ubuntu 21.10

### What version of Gnome are you using?

40

corecoding, 2022-03-25:
The latest Vitals changed the keys under the hood used to keep track of Temp, Fan and Volt sensors shown in the top bar. The old behavior was that clicking off (or on) any sensor would remove the missing "..." icons. But the function that removes missing sensors was moved to the Refresh button. If you tap the Refresh button, it should remove missing "..." sensors and you can then re-click the ones you want to be visible.

heynnema, 2022-03-26:
But your instructions don't work any more. They did when I first did the recent upgrade, but now, the "..." icon returns, I hit refresh, re-select the desired temp sensor, and then later it reverts again.

Where are the prefs stored? I could delete them and try again.

corecoding, 2022-03-27:
I actually don't know where the preferences are stored, gnome handles it for you..

Expanding on the statement above, I am wondering if it is a device, such as an AC adapter (assuming you are on a laptop), a USB device, etc that is plugged in (or unplugged) that is causing a shuffle in the keys that I am now using. Can you confirm whether it is something you are doing (like a reboot) or plugging into the computer, or is it just random when "..." appears?

corecoding, 2022-03-27:
One more thought. Is the sensor that turns to "..." still in the Vitals menu dropdown but possibly greyed out?

heynnema, 2022-03-27:
Ah, that made me think.

My temp sensors flip around at boot time. Here's a commented out excerpt from my thermald config file:

               <ThermalSensor>
                        <Type>iwlwifi_1</Type>
                        <Path>/sys/class/thermal/thermal_zone1/</Path>
                        <AsyncCapable>0</AsyncCapable>
                </ThermalSensor>
                <ThermalSensor>
                        <Type>x86_pkg_temp</Type>
                        <Path>/sys/class/thermal/thermal_zone2/</Path>
                        <AsyncCapable>1</AsyncCapable>

The first is normally the hard path for the modem card temp sensor. The second is normally the hard path for the CPU temp sensor. But sometimes, at boot, the hard paths flip between the two devices, modem vs CPU.

I had to modify my thermals config file to remove the references to the hard paths.

The two paths look like this:

$ ll /sys/class/thermal/thermal_zone1/

total 0
drwxr-xr-x  4 root root    0 Mar 25 05:53 ./
drwxr-xr-x 20 root root    0 Mar 24 22:53 ../
-r--r--r--  1 root root 4096 Mar 27 06:07 available_policies
--w-------  1 root root 4096 Mar 27 06:07 emul_temp
drwxr-xr-x  3 root root    0 Mar 25 05:53 hwmon2/
-rw-r--r--  1 root root 4096 Mar 27 06:07 integral_cutoff
-rw-r--r--  1 root root 4096 Mar 27 06:07 k_d
-rw-r--r--  1 root root 4096 Mar 27 06:07 k_i
-rw-r--r--  1 root root 4096 Mar 27 06:07 k_po
-rw-r--r--  1 root root 4096 Mar 27 06:07 k_pu
-rw-r--r--  1 root root 4096 Mar 27 06:07 mode
-rw-r--r--  1 root root 4096 Mar 27 06:07 offset
-rw-r--r--  1 root root 4096 Mar 27 06:07 policy
drwxr-xr-x  2 root root    0 Mar 26 06:20 power/
-rw-r--r--  1 root root 4096 Mar 27 06:07 slope
lrwxrwxrwx  1 root root    0 Mar 25 05:53 subsystem -> ../../../../class/thermal/
-rw-r--r--  1 root root 4096 Mar 27 06:07 sustainable_power
-r--r--r--  1 root root 4096 Mar 25 05:53 temp
-rw-r--r--  1 root root 4096 Mar 25 05:53 trip_point_0_temp
-r--r--r--  1 root root 4096 Mar 25 05:53 trip_point_0_type
-rw-r--r--  1 root root 4096 Mar 25 05:53 trip_point_1_temp
-r--r--r--  1 root root 4096 Mar 25 05:53 trip_point_1_type
-rw-r--r--  1 root root 4096 Mar 25 05:53 trip_point_2_temp
-r--r--r--  1 root root 4096 Mar 25 05:53 trip_point_2_type
-rw-r--r--  1 root root 4096 Mar 25 05:53 trip_point_3_temp
-r--r--r--  1 root root 4096 Mar 25 05:53 trip_point_3_type
-rw-r--r--  1 root root 4096 Mar 25 05:53 trip_point_4_temp
-r--r--r--  1 root root 4096 Mar 25 05:53 trip_point_4_type
-rw-r--r--  1 root root 4096 Mar 25 05:53 trip_point_5_temp
-r--r--r--  1 root root 4096 Mar 25 05:53 trip_point_5_type
-rw-r--r--  1 root root 4096 Mar 25 05:53 trip_point_6_temp
-r--r--r--  1 root root 4096 Mar 25 05:53 trip_point_6_type
-rw-r--r--  1 root root 4096 Mar 25 05:53 trip_point_7_temp
-r--r--r--  1 root root 4096 Mar 25 05:53 trip_point_7_type
-r--r--r--  1 root root 4096 Mar 25 05:53 type
-rw-r--r--  1 root root 4096 Mar 25 05:53 uevent

$ ll /sys/class/thermal/thermal_zone2/

total 0
drwxr-xr-x  3 root root    0 Mar 25 05:53 ./
drwxr-xr-x 20 root root    0 Mar 24 22:53 ../
-r--r--r--  1 root root 4096 Mar 27 06:07 available_policies
--w-------  1 root root 4096 Mar 27 06:07 emul_temp
-rw-r--r--  1 root root 4096 Mar 27 06:07 integral_cutoff
-rw-r--r--  1 root root 4096 Mar 27 06:07 k_d
-rw-r--r--  1 root root 4096 Mar 27 06:07 k_i
-rw-r--r--  1 root root 4096 Mar 27 06:07 k_po
-rw-r--r--  1 root root 4096 Mar 27 06:07 k_pu
-rw-r--r--  1 root root 4096 Mar 27 06:07 mode
-rw-r--r--  1 root root 4096 Mar 27 06:07 offset
-rw-r--r--  1 root root 4096 Mar 27 06:07 policy
drwxr-xr-x  2 root root    0 Mar 26 06:20 power/
-rw-r--r--  1 root root 4096 Mar 27 06:07 slope
lrwxrwxrwx  1 root root    0 Mar 25 05:53 subsystem -> ../../../../class/thermal/
-rw-r--r--  1 root root 4096 Mar 27 06:07 sustainable_power
-r--r--r--  1 root root 4096 Mar 25 05:53 temp
-rw-r--r--  1 root root 4096 Mar 25 05:53 trip_point_0_temp
-r--r--r--  1 root root 4096 Mar 25 05:53 trip_point_0_type
-rw-r--r--  1 root root 4096 Mar 25 05:53 trip_point_1_temp
-r--r--r--  1 root root 4096 Mar 25 05:53 trip_point_1_type
-r--r--r--  1 root root 4096 Mar 25 05:53 type
-rw-r--r--  1 root root 4096 Mar 25 05:53 uevent

And you can see that the "temp" path is different, depending on how it decides at boot time.

corecoding, 2022-04-02:
This post was very helpful! Thank you!

It appears that #245 is a duplicate. Normally I would close the newer ticket, but more people chimed in over there so I am going to close this issue.

In #245, I have asked for some beta testing help - I reverted some code to go back to the old behavior. If you could chime in over there and help beta test I would greatly appreciate it.

closed 2022-04-02 by corecoding