corecoding/Vitals #245: Dangling temp. icon

GregoryOrciuch, 2022-03-30:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Kind of week ago I've updated Vitals to latest version on my Fedora35 thru Firefox's extension plugin.
After that action and reboot one temperature icon is dangling, like not showing any temp. Another choice of temp point is showing well but idk how to get rid of those dummy/dangling .
Attaching screenshot

### Steps to reproduce the unexpected behavior.

I just upgrade :/

### What distribution and version of Linux are you using?

Fedora 35

### What version of Gnome are you using?

41.5

GregoryOrciuch, 2022-03-30:
[image]

mvaled, 2022-03-31:
I'm getting the same left-over icons.  I have two already:

[image]

skgsergio, 2022-04-01:
Same issue here, as a workaround for removing them you can do the following:

1. Disable extension
2. Open a terminal and manually change the list of sensors:
    ```
    $ dconf read /org/gnome/shell/extensions/vitals/hot-sensors
    ['_memory_usage_', '_system_load_1m_', '_network-rx_wlp0s20f3_rx_', '_network-tx_wlp0s20f3_tx_', '_temperature_hwmon3temp1_', '_temperature_hwmon4temp1_']
    
    $ dconf write /org/gnome/shell/extensions/vitals/hot-sensors "['_memory_usage_', '_system_load_1m_', '_network-rx_wlp0s20f3_rx_', '_network-tx_wlp0s20f3_tx_']" 
    
    $ dconf read /org/gnome/shell/extensions/vitals/hot-sensors
    ['_memory_usage_', '_system_load_1m_', '_network-rx_wlp0s20f3_rx_', '_network-tx_wlp0s20f3_tx_']
    ```
4. Enable the extension

corecoding, 2022-04-01:
Do these seem to happen after a reboot?

corecoding, 2022-04-02:
I have reverted some code to operate in the previous fashion. I will need some beta testers help. The instructions for Beta testing are here: https://github.com/corecoding/Vitals/blob/develop/README.md#beta-testing

Please note that reverting the code will cause "..." to appear again, but once you toggle a sensor on or off, it should clear up and not come back.

corecoding, 2022-04-08:
Has anyone been able to test this develop build?

skgsergio, 2022-04-08:
@corecoding I've installed it a couple of days ago but since I removed the missing sensors manually with `dconf` I couldn't reproduce the bug.

I've reproduced it manually adding back the missing sensors with dconf and re-logged in my gnome session to make it effective:

```
dconf write /org/gnome/shell/extensions/vitals/hot-sensors "['_memory_usage_', '_system_load_1m_', '_network-rx_wlp0s20f3_rx_', '_network-tx_wlp0s20f3_tx_', '_temperature_hwmon3temp1_', '_temperature_hwmon4temp1_']"
```

At this point, I had two temperature icons with "..." and I thought it wasn't working but then when selected again the one I wanted the ones with "..." disappeared. So I guess the cleanup is not performed on extension load but when you change the hot sensors.

If that is the expected behavior then it is working :)

corecoding, 2022-04-08:
Thank you. As I've told other people, if I were to remove ... sensors automatically, I'd be getting just as many bug reports about sensors disappearing. I for one have a laptop with a Hybrid graphics, and my GPU isn't always used. When I switch from an external display to the laptop screen, it switches from Nvidia to Intel. If I have a GPU sensor selected, and if I were to remove it when switching to laptop mode, people would be like "Why is the sensor going away???" It's a hard problem to solve intuitively.

skgsergio, 2022-04-08:
I don't know how much work is but maybe a list in the config where you can remove (and maybe re-order) sensors?

jose1711, 2022-04-10:
I had a similar issue but for me there was no icon at all - there was just an empty space you could click and bring up the context menu. Following the steps in https://github.com/corecoding/Vitals/issues/245#issuecomment-1085631045 did resolve the problem:
```bash
$ gnome-extensions disable Vitals@CoreCoding.com
$ dconf read /org/gnome/shell/extensions/vitals/hot-sensors
@as []
$ dconf write /org/gnome/shell/extensions/vitals/hot-sensors "['_memory_usage_']"
$ dconf read /org/gnome/shell/extensions/vitals/hot-sensors
['_memory_usage_']
$ gnome-extensions enable Vitals@CoreCoding.com
```
It's interesting that now even if all sensors in the tray are disabled the `dconf` value is `['_default_icon_']` (not `@as[]` as before).

corecoding, 2022-04-14:
> I don't know how much work is but maybe a list in the config where you can remove (and maybe re-order) sensors? I mean the issue now is that in your scenario if you change a sensor while not using the GPU it is going to remove your sensor.

Take a look at https://github.com/corecoding/Vitals/issues/93

corecoding, 2022-04-14:
> I had a similar issue but for me there was no icon at all - there was just an empty space you could click and bring up the context menu. Following the steps in [#245 (comment)](https://github.com/corecoding/Vitals/issues/245#issuecomment-1085631045) did resolve the problem:
> 
> ```shell
> $ gnome-extensions disable Vitals@CoreCoding.com
> $ dconf read /org/gnome/shell/extensions/vitals/hot-sensors
> @as []
> $ dconf write /org/gnome/shell/extensions/vitals/hot-sensors "['_memory_usage_']"
> $ dconf read /org/gnome/shell/extensions/vitals/hot-sensors
> ['_memory_usage_']
> $ gnome-extensions enable Vitals@CoreCoding.com
> ```
> 
> It's interesting that now even if all sensors in the tray are disabled the `dconf` value is `['_default_icon_']` (not `@as[]` as before).

The intent is that hot-sensors would never be []. If no sensors are detected, it would be set to `['_default_icon_']`. If you can find a way to duplicate the hot-sensors being set to [], I think it would be valuable input for #243

corecoding, 2022-04-14:
Fixed version has been released. Please know that it may show ... one final time, toggling a sensor will clear ... entries and you can then toggle the sensors back. Going forward it should work properly.

closed 2022-04-14 by corecoding

GregoryOrciuch, 2022-04-14:
@corecoding Than you, all good now!