corecoding/Vitals #246: Data read(write)/sec is not display

shungo27, 2022-04-02:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Storage > Data read(write)/sec is not display.
[image]

### Steps to reproduce the unexpected behavior.

1. install Vitals
$ sudo apt install gir1.2-gtop-2.0 lm-sensors
https://extensions.gnome.org/extension/1460/vitals/

* raspberry pi 4b 8gb
* SSD and HDD are connected to USB

### What distribution and version of Linux are you using?

raspberry pi OS 64bit Lite bullseye

### What version of Gnome are you using?

3.38

corecoding, 2022-04-05:
Vitals reads /proc/diskstats and /proc/mounts - only when both of those device blocks can be matched does the read/write show. When using LVM, ZFS, etc this feature is not supported.

closed 2022-04-05 by corecoding

shungo27, 2022-04-05:
thank you for information. I sloved this issue.

sensors.js
```
- this._storageDevice = '';
- this._findStorageDevice();
+ // this._storageDevice = '';
+ // this._findStorageDevice();

- if ('/dev/' + loadArray[2] == this._storageDevice) {
+ if (loadArray[2].match(/^sd.$/)) {
```

On my system all storage starts with "sd" so this was the solution.