corecoding/Vitals #247: CPU temperature, switching to wifi card temperature after Suspense/Reboot

AdsonCicilioti, 2022-04-09:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Selected any CPU temperature (Package/Core 1/2/3/.../Average), after system suspend, reboot or power on, it changes to Wifi card temperature. This behavior occurs randomly.

### Steps to reproduce the unexpected behavior.

1. Select any CPU temperature
2. Await system enter on suspension, reboot or power on
3. the chosen temp is switched to Wi-Fi card temp.

### What distribution and version of Linux are you using?

Arch Linux, Kernel 5.17 zen.

### What version of Gnome are you using?

41, 42 (behavior seen in both)

corecoding, 2022-04-09:
Although it isn't obvious, this is a duplicate of #245. Please see the second to last post and if you can try the development version and let me know if it is fixed, I'd appreciate it.

closed 2022-04-09 by corecoding

AdsonCicilioti, 2022-04-10:
@corecoding , I've switch to `develop` branch and the problem seem is fixed.

corecoding, 2022-04-10:
Thank you! I will likely release the update today then. I had another person say it was also fixed. I am confident because I reverted the code to the way it used to work, instead of making changes to the newer code.