corecoding/Vitals #250: Vitals causing gnome-shell to 100% CPU and locks up computer

heynnema, 2022-04-13:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Randomly, my computer is locking up. Looking at "top" I can see that gnome-shell is at 100%. Reboot required. I don't know how to tell the version of Vitals, but it's the current version... the one with the "Icon ..." problem. This is a new problem.

### Steps to reproduce the unexpected behavior.

Nothing specific required to trigger the problem.

### Relevant log output

```shell
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_parser_new_from_buf: assertion 'a_buf && a_len' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_declaration_parse_list_from_buf: assertion 'parser' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_parser_new_from_buf: assertion 'a_buf && a_len' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_declaration_parse_list_from_buf: assertion 'parser' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_parser_new_from_buf: assertion 'a_buf && a_len' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_declaration_parse_list_from_buf: assertion 'parser' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_parser_new_from_buf: assertion 'a_buf && a_len' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_declaration_parse_list_from_buf: assertion 'parser' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_parser_new_from_buf: assertion 'a_buf && a_len' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_declaration_parse_list_from_buf: assertion 'parser' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: DING: Gjs-Message: 09:14:59.602: JS LOG: Error connecting to Nautilus#012GNOME nautilus 40.2
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_parser_new_from_buf: assertion 'a_buf && a_len' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: cr_declaration_parse_list_from_buf: assertion 'parser' failed
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: Object St.Label (0x5650ca51a430), has been already deallocated â impossible to set any property on it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: == Stack trace for context 0x5650c8a051b0 ==
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #0   5650cc5fb378 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/menuItem.js:53 (3d7e330286f0 @ 18)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #1   5650cc5fb2c8 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:376 (3d7e3301f150 @ 259)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #2   7ffcf3cc66e0 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:531 (3d7e3300ce70 @ 283)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #3   7ffcf3cc6f00 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:470 (3d7e3301ff60 @ 240)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #4   5650cc5fb208 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:367 (3d7e330252e0 @ 184)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: == Stack trace for context 0x5650c8a051b0 ==
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #0   5650cc5fb378 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/menuItem.js:53 (3d7e330286f0 @ 18)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #1   5650cc5fb2c8 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:376 (3d7e3301f150 @ 259)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #2   7ffcf3cc66e0 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:531 (3d7e3300ce70 @ 283)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #3   7ffcf3cc6f00 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:470 (3d7e3301ff60 @ 240)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #4   5650cc5fb208 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:367 (3d7e330252e0 @ 184)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: == Stack trace for context 0x5650c8a051b0 ==
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #0   5650cc5fb378 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/menuItem.js:53 (3d7e330286f0 @ 18)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #1   5650cc5fb2c8 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:376 (3d7e3301f150 @ 259)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #2   7ffcf3cc66e0 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:531 (3d7e3300ce70 @ 283)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #3   7ffcf3cc6f00 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:470 (3d7e3301ff60 @ 240)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #4   5650cc5fb208 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:367 (3d7e330252e0 @ 184)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: Object St.Label (0x5650cda85430), has been already deallocated â impossible to set any property on it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: == Stack trace for context 0x5650c8a051b0 ==
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #0   5650cc5fb378 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/menuItem.js:53 (3d7e330286f0 @ 18)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #1   5650cc5fb2c8 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:376 (3d7e3301f150 @ 259)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #2   7ffcf3cc66e0 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:531 (3d7e3300ce70 @ 283)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #3   7ffcf3cc6f00 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:470 (3d7e3301ff60 @ 240)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #4   5650cc5fb208 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:367 (3d7e330252e0 @ 184)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: == Stack trace for context 0x5650c8a051b0 ==
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #0   5650cc5fb378 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/menuItem.js:53 (3d7e330286f0 @ 18)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #1   5650cc5fb2c8 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:376 (3d7e3301f150 @ 259)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #2   7ffcf3cc66e0 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:531 (3d7e3300ce70 @ 283)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #3   7ffcf3cc6f00 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:470 (3d7e3301ff60 @ 240)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #4   5650cc5fb208 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:367 (3d7e330252e0 @ 184)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: == Stack trace for context 0x5650c8a051b0 ==
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #0   5650cc5fb2c8 i   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:381 (3d7e3301f150 @ 380)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #1   7ffcf3cc66e0 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:531 (3d7e3300ce70 @ 283)
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: #2   7ffcf3cc6f00 b   /home/al/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:470 (3d7e3301ff60 @ 240)
[... cut ...] from C code using something such as destroy(), dispose(), or remove() vfuncs.
Apr 12 09:51:04 al-Inspiron-7700-AIO gnome-shell[3698]: Object St.Label (0x5650d09e8810), has been already deallocated â impossible to set any property on it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
```

### What distribution and version of Linux are you using?

Ubuntu 21.10

### What version of Gnome are you using?

40.4.0

corecoding, 2022-04-14:
What is your "Seconds between updates" set to?

heynnema, 2022-04-14:
5 seconds

heynnema, 2022-04-30:
Status please...

corecoding, 2022-05-01:
There isn't a lot I can do without having access to the same hardware / os installation that you have. https://github.com/corecoding/Vitals/issues/150 This is a related conversation where this person's Kernel caused a sensor poll to use high CPU usage.

heynnema, 2022-05-01:
This problem is **very** intermittent. But it only started happening at the same time the sensors (icon...) problem showed up. And it seems to effect gnome-shell going nuts and hanging the system. Was any code changed in the gnome-shell code, when you made changes that caused the (icon...) problem? Prior to the last update, it all worked fine.

corecoding, 2022-06-21:
Making note - likely a duplicate of #260

heynnema, 2022-06-23:
@corecoding The development version still has the same problem. Lots of stack traces again.

corecoding, 2022-06-26:
@heynnema sorry to be a bother, I just realized I never pushed the change that I think fixed your issue. If you still have the develop branch can you do a `git pull` and log out of your shell and back in? 

If it still doesn't work, can I get one more stack trace, as the line numbers have changed

corecoding, 2022-07-04:
Closing as a duplicate of #260. Still would like confirmation that the develop branch fixes the lag.

heynnema, 2022-07-04:
@corecoding The latest develop version has no fan data.

corecoding, 2022-07-10:
Reopening because author of #260 switched away from Gnome.

corecoding, 2022-12-12:
Can anyone try the latest v58 of Vitals to see if the issue persists?

corecoding, 2023-02-01:
Closing due to inactivity.

closed 2023-02-01 by corecoding