corecoding/Vitals #252: data is replaced by points in the bar on ubuntu 22.04

maravento, 2022-04-25:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

should display data instead of ellipses

[image]

### Steps to reproduce the unexpected behavior.

see error

### What distribution and version of Linux are you using?

ubuntu 22.04

### What version of Gnome are you using?

42

pjadczak, 2022-04-26:
I have the sam
[image]
e

rungis86, 2022-04-29:
Idem on Fedora 35 Gnome 41.5

sebastienserre, 2022-04-30:
Same issue here

corecoding, 2022-04-30:
Are you all using English or another language?

pjadczak, 2022-05-01:
No, I use Polish language.

rungis86, 2022-05-01:
I'm using french.

sebastienserre, 2022-05-01:
French too

e7d, 2022-05-03:
Hello. I had the same problem a few weeks ago on my Manjaro, that I solved by clicking the refresh button, on the bottom left.
[image]

Hope it does the trick for you too!

corecoding, 2022-05-04:
I have identified the issue. The issue was that for non English installs, it was trying to resolve Memoria instead of Memory (for example). The fix has been uploaded and should be available within a few days.

corecoding, 2022-05-04:
New version just released.

closed 2022-05-04 by corecoding