corecoding/Vitals #253: Not starting anymore after 22.04 upgrade

raikoug, 2022-04-28:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Vitals is not showing after upgrading to 22.04
System info:
- OS: Pop!_OS 22.04 LTS x86_64
- Kernel: 5.16.19-76051619-generic
- DE: GNOME 42.0
- GNOME Shell 42.0
- Vitals: v53

It was working soon before the upgrade.  
I tried to uninstall and reinstall. But switching the button to ON con gnome-extension just go back to OFF after reloading the page.  
Setting page works.
I tried to uninstall and reinstall via git clone, nothing.  
The Logs I found are not telling the problem.

```
sudo journalctl /usr/bin/gnome-shell -f
apr 28 19:32:58 popkoug gnome-shell[18393]: Thank you, please carry on
apr 28 19:32:58 popkoug gnome-shell[18393]: [VERTICAL_OVERVIEW] enabled
apr 28 19:32:59 popkoug gnome-shell[18393]: Error looking up permission: GDBus.Error:org.freedesktop.portal.Error.NotFound: No entry for geolocation
apr 28 19:33:00 popkoug gnome-shell[18393]: Window manager warning: Overwriting existing binding of keysym 6f with keysym 6f (keycode 20).
apr 28 19:33:01 popkoug gnome-shell[18393]: GNOME Shell started at Thu Apr 28 2022 19:32:58 GMT+0200 (CEST)
apr 28 19:33:01 popkoug gnome-shell[18393]: Registering session with GDM
apr 28 19:33:01 popkoug gnome-shell[18393]: DING: Detected async api for thumbnails
apr 28 19:33:01 popkoug gnome-shell[18393]: DING: GNOME nautilus 42.0
apr 28 19:33:07 popkoug gnome-shell[18393]: Window manager warning: Overwriting existing binding of keysym 6f with keysym 6f (keycode 20).
apr 28 19:33:59 popkoug ubuntu-appindicators@ubuntu.com[18393]: unable to update icon for MEGAsync
```

Other gnome-shell extensions are working fine (cosmic dock/workspace, pop cosmic/shell, System76 Power , ubuntu-appindicators and desktop-icons-ng-ding)

### Steps to reproduce the unexpected behavior.

1. Install Vitals
2. Upgrade system to 22.04

### Relevant log output

```shell
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/LICENSE
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/helpers/
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/helpers/file.js
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/helpers/polyfills.js
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/battery-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/fan-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/voltage-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/storage-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/network-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/network-download-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/temperature-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/network-upload-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/system-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/cpu-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/icons/memory-symbolic.svg
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/fi_FI/
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/fi_FI/LC_MESSAGES/
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/fi_FI/LC_MESSAGES/vitals.mo
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/pt/
[... 32 lines cut ...]
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/ca/
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/ca/LC_MESSAGES/
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/ca/LC_MESSAGES/vitals.mo
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/nl/
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/nl/LC_MESSAGES/
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/locale/nl/LC_MESSAGES/vitals.mo
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/menuItem.js
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/prefs.js
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/prefs.legacy.ui
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/prefs.ui
apr 28 19:27:46 popkoug gnome-shell[16877]:    creating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/schemas/
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/schemas/gschemas.compiled
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/stylesheet.css
apr 28 19:27:46 popkoug gnome-shell[16877]:   inflating: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/values.js
apr 28 19:27:46 popkoug gnome-shell[16877]:  extracting: /home/raikoug/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/metadata.json
```

### What distribution and version of Linux are you using?

Pop_OS 22.04

### What version of Gnome are you using?

42

corecoding, 2022-04-30:
https://github.com/corecoding/Vitals/wiki/Broken-after-upgrade
`The new version is then loaded on your next session`

raikoug, 2022-04-30:
This is not my case.
There is no error `Error: A different version was loaded previously. You need to log out for changes to take effect.`
Every other extension, as I mentioned, is working.
Why closing without reading? I already checked the wiki.
I already powered off my laptop several times and logged out-in.

corecoding, 2022-05-01:
Don't assume I didn't read. I closed the issue because I get a report like this about 4x a month. In all other cases, logging out and logging in fixes the issue.  Unfortunately, the output you gave above doesn't give any insight. I run POP myself and v53 is working great.

Take a look at /var/log/syslog and see if you see any Vitals entries that are a stack trace.

raikoug, 2022-05-02:
Sorry for assuming, but I pasted the logs of the wiki to demonstrate that's not my case. 
I also said all others gnome extensions are working well.

By the way, there is nothing in the syslog about Vitals.
This is the log of "`alt f2 - r`"

```
journalctl  -f | grep gnome-shell
mag 02 11:44:09 popkoug gnome-shell[17610]: Window manager warning: Failed to start restart helper: Failed to execute child process “/usr/libexec/mutter-restart-helper” (No such file or directory)
mag 02 11:44:09 popkoug gnome-shell[17610]: clutter_actor_iter_next: assertion 'ri->age == ri->root->priv->age' failed
mag 02 11:44:09 popkoug gnome-shell[17610]: Object .MetaWindowActorX11 (0x55c1fd39a370), has been already disposed — impossible to get any property from it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
mag 02 11:44:09 popkoug gnome-shell[17610]: == Stack trace for context 0x55c1f5af3190 ==
mag 02 11:44:09 popkoug gnome-shell[17610]: #0   55c1f734da28 i   resource:///org/gnome/shell/ui/windowManager.js:1542 (352220829c90 @ 23)
mag 02 11:44:09 popkoug gnome-shell[17610]: #1   7ffe9618d5c0 b   self-hosted:1181 (6de5b1b0a10 @ 454)
mag 02 11:44:09 popkoug gnome-shell[17610]: #2   55c1f734d9a8 i   resource:///org/gnome/shell/ui/main.js:260 (6de5b1ef150 @ 12)
mag 02 11:44:09 popkoug gnome-shell[17610]: JS ERROR: TypeError: window is null
mag 02 11:44:09 popkoug kernel: gnome-shell[17610]: segfault at 80 ip 00007f4c14d71074 sp 00007ffe9618e1d8 error 4 in libmutter-10.so.0.0.0[7f4c14cce000+142000]
mag 02 11:44:10 popkoug gnome-shell[18385]: Failed to set RT scheduler: Operation not permitted
mag 02 11:44:10 popkoug gnome-shell[18385]: Running GNOME Shell (using mutter 42.0) as a X11 window and compositing manager
mag 02 11:44:10 popkoug gnome-shell[18385]: Enabling experimental feature 'x11-randr-fractional-scaling'
mag 02 11:44:10 popkoug gnome-shell[18385]: ATK Bridge is disabled but a11y has already been enabled.
mag 02 11:44:10 popkoug gnome-shell[18385]: Unset XDG_SESSION_ID, getCurrentSessionProxy() called outside a user session. Asking logind directly.
mag 02 11:44:10 popkoug gnome-shell[18385]: Will monitor session 3
mag 02 11:44:10 popkoug dbus-daemon[1174]: [system] Activating via systemd: service name='org.freedesktop.GeoClue2' unit='geoclue.service' requested by ':1.259' (uid=1000 pid=18385 comm="/usr/bin/gnome-shell ")
mag 02 11:44:10 popkoug polkitd(authority=local)[1190]: Registered Authentication Agent for unix-session:3 (system bus name :1.259 [/usr/bin/gnome-shell], object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8)
mag 02 11:44:10 popkoug gnome-shell[18385]: Telepathy is not available, chat integration will be disabled.
mag 02 11:44:11 popkoug gnome-shell[18385]: Failed to import DBusMenu, quicklists are not avaialble: Error: Requiring Dbusmenu, version none: Typelib file for namespace 'Dbusmenu' (any version) not found
mag 02 11:44:11 popkoug dbus-daemon[2951]: [session uid=1000 pid=2951] Activating service name='org.freedesktop.FileManager1' requested by ':1.215' (uid=1000 pid=18385 comm="/usr/bin/gnome-shell ")
mag 02 11:44:11 popkoug gnome-shell[18385]: gnome-shell-extension-system76-power: power profile was set: 'Balanced'
mag 02 11:44:11 popkoug gnome-shell[18385]: [VERTICAL-OVERVIEW] starting overrides
mag 02 11:44:11 popkoug gnome-shell[18385]: Some code accessed the property 'SecondaryMonitorDisplay' on the module 'workspacesView'. That property was defined with 'let' or 'const' inside the module. This was previously supported, but is not correct according to the ES6 standard. Any symbols to be exported from a module must be defined with 'var'. The property access will work as previously for the time being, but please fix your code anyway.
mag 02 11:44:11 popkoug gnome-shell[18385]: You may see an error below,
mag 02 11:44:11 popkoug gnome-shell[18385]: Thank you, please carry on
mag 02 11:44:11 popkoug gnome-shell[18385]: [VERTICAL_OVERVIEW] enabled
mag 02 11:44:11 popkoug dbus-daemon[2951]: [session uid=1000 pid=2951] Activating service name='org.gnome.Shell.Notifications' requested by ':1.215' (uid=1000 pid=18385 comm="/usr/bin/gnome-shell ")
mag 02 11:44:11 popkoug gnome-shell[18385]: Error looking up permission: GDBus.Error:org.freedesktop.portal.Error.NotFound: No entry for geolocation
mag 02 11:44:11 popkoug gnome-shell[18385]: Unable to mount volume HUAWEI: Gio.IOErrorEnum: Location is already mounted
mag 02 11:44:13 popkoug dbus-daemon[2951]: [session uid=1000 pid=2951] Activating service name='org.gnome.Shell.HotplugSniffer' requested by ':1.215' (uid=1000 pid=18385 comm="/usr/bin/gnome-shell ")
mag 02 11:44:13 popkoug gnome-shell[18385]: Window manager warning: Overwriting existing binding of keysym 6f with keysym 6f (keycode 20).
mag 02 11:44:13 popkoug gnome-shell[18385]: Can't update stage views actor <unnamed>[<MetaWindowGroup>:0x55ca2f5fe370] is on because it needs an allocation.
mag 02 11:44:13 popkoug gnome-shell[18385]: Can't update stage views actor <unnamed>[<MetaWindowActorX11>:0x55ca321c2f00] is on because it needs an allocation.
mag 02 11:44:13 popkoug gnome-shell[18385]: Can't update stage views actor <unnamed>[<MetaSurfaceActorX11>:0x55ca321c6dc0] is on because it needs an allocation.
mag 02 11:44:13 popkoug gnome-shell[18385]: GNOME Shell started at Mon May 02 2022 11:44:11 GMT+0200 (CEST)
[... cut ...]
mag 02 11:44:14 popkoug gnome-shell[18385]: Can't update stage views actor <unnamed>[<MetaSurfaceActorX11>:0x55ca321c6dc0] is on because it needs an allocation.
mag 02 11:44:14 popkoug dbus-daemon[2951]: [session uid=1000 pid=2951] Activating service name='org.gnome.ArchiveManager1' requested by ':1.220' (uid=1000 pid=18537 comm="gjs /usr/share/gnome-shell/extensions/ding@rasters")
mag 02 11:44:14 popkoug gnome-shell[18385]: DING: Detected async api for thumbnails
mag 02 11:44:14 popkoug gnome-shell[18385]: DING: GNOME nautilus 42.0
```

I don't see anything related to Vitals but the errors are there.. dunno.

corecoding, 2022-05-05:
Is there any mention of Vitals in `/var/log/syslog` ? Also, what is the output of this command? `dconf read /org/gnome/shell/extensions/vitals/hot-sensors`

corecoding, 2022-06-16:
Closing this ticket - no response for a while. No one else has reported this issue, so assuming it is fixed.

closed 2022-06-16 by corecoding