corecoding/Vitals #255: Weird round borders in preferences

dasnoopy, 2022-05-05:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Vital extensions ver. 54: 

Noticed recently, but all sensors preferences  window has round  borders inside the windows itself.

Theme is  Gnome v42 default **adwaita**  with close button on the left..

an example:
[image]

regards

### Steps to reproduce the unexpected behavior.

1) open the settings windows
2) open any of  sensor preference window
3) look corners inside the window

### What distribution and version of Linux are you using?

archlinux 64bit

### What version of Gnome are you using?

gnome 42.1

corecoding, 2022-05-13:
I have a fresh copy of Arch with the latest vitals version. It looks fine to me.
[image]
 
The preferences UI is generated from prefs.ui - it controls the elements by not the pixels/styling.

dasnoopy, 2022-05-13:
opss.. forgotten that I was playing with  ./config/gtk-4/gtk.css file... 

this "setting"  is the culprit
```
/* less rounded windows border */
window { border-radius: 8px; }
```

sorry and thanks for your time

closed 2022-05-13 by dasnoopy