corecoding/Vitals #256: CPU PERCENTAGE GONE

gnolooo, 2022-05-07:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

After the latest update, I can't set as a pinnable icon the average CPU percentage usage. I can only pin the Bogomips value, the boost, the cache and the single cores (unlike for example in RAM, where the average percentage is present).

This is what I see:
[image]

### Steps to reproduce the unexpected behavior.

1. Go to top panel
2. Click the CPU section

### What distribution and version of Linux are you using?

Arch Linux

### What version of Gnome are you using?

42

corecoding, 2022-05-07:
When you say you can't "set as pinnable" what do you mean? When you click Media under Processor, what exactly happens?

gnolooo, 2022-05-07:
When I say "set as pinnable" I mean clickable to pin it on the dash. There's no "average CPU usage" section (except for the sub-menu one) that I can click to pin to the dash, as I can do for example with RAM instead.
Basically I would like to pin this voice (which is average CPU usage) on the dash:
[image]
but I can't do so since the voice is missing in the submenu.
I'm pretty sure this is a bug because it was there until last update, and I don't think you remove it. But I've tried uninstalling it, clearing cache etc, and nothing worked.

For reference, this is what I used to configure Vitals like (sorry for bad quality, it's an old screenshot):
[image]

P.S.: the CPU icon is different just because I changed it, but afaik that shouldn't affect the extension behavior itself.

gnolooo, 2022-05-07:
I am so dumb, I didn't see that one was a scrolling menu. The voice i was looking for was just under. Sorry for wasting your time.

closed 2022-05-07 by gnolooo