corecoding/Vitals #257: Equally named sensors are merged ( reopen of #240)

maxaalderink, 2022-05-12:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Reopened issue #240.
Since the latest update (version 53) I have noticed that I can only find one out of two NVME SSD in the temperature dropdown.
Initially I thought it only showed the drive I'm currently using as the second one is for a secondary OS.
But since lm-sensors en Psensor is still showing both I think the problem is related to Vitals.
Additionally I've seen the the temperature jump 20+C on occasions, if I had to guess that is because it sometimes reads and displays the sensor from the secondary NVME which is much cooler because it's not being used actively.

The drives uses are a Samsung 980 Pro and a Samsung 970 Evo+, for both lm-sensors detects 3 senors named Sensor 1, Sensor 2 and Composite.

### Steps to reproduce the unexpected behavior.

1. Install 2 devices that present equally named sensors to lm-sensors.
2. Run sensors-detect.
3. Run sensors to see if both devices are detected (in my case nvme-pci-0400 & nvme-pci-0100) with equally named sensors.
4. Notice that only one set of sensors can be found in Vitals.

### What distribution and version of Linux are you using?

Pop!_OS 21.10

### What version of Gnome are you using?

40.4.0

corecoding, 2022-05-13:
At first I was thinking "no way" but after some research, I found out what happened. The code in #240 changed how sensors were handled by using a key value derived from the /sys/class paths. Some users were having issues where every boot, the structure of this directory would change and sensors in the top bar were flopped.

I reverted to the previous behavior, and it looks like I missed the part about appending ' 2' at the end. My goal is to clean up this code, but for now, the develop branch should be utilizing the old code.

Would you mind checking out the develop branch for me and see if the issue is fixed? You'll have to log out and log back in after running these commands: https://github.com/corecoding/Vitals#1-remove-existing-copy-of-vitals

maxaalderink, 2022-05-13:
I'll try to try it out this weekend

corecoding, 2022-06-03:
Just checking in to see if you have had time to test the develop branch?

maxaalderink, 2022-06-14:
I'm very sorry for the lack of response, something came up and I forgot about it.
I'm going to try to test it tomorrow

corecoding, 2022-06-25:
Thanks. Please let me know so I can release an update :)

corecoding, 2022-07-10:
Closing due to no response. Please feel free to bring up the issue again after the next release.

closed 2022-07-10 by corecoding