corecoding/Vitals #260: Vitals produces system lag when checking sensor/system status

ondrejsource, 2022-05-18:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

I recently noticed my cursor in GNOME lagging every 3 seconds or so. At first, I thought it was a hardware issue (because Linux and drivers, you know), but then I realized the lag was happening in consistent intervals. I tried to guess how long those intervals were, and then I realized they were long exactly 3 seconds - the interval I set in Vitals. I went into Vitals' settings and set the interval to 5 seconds, and sure enough, the interval increased to 5 seconds. I checked the system monitor, but Vitals wasn't really using a lot of resources (as far as I can tell). So I have disabled Vitals on my system for now. My laptop isn't a piece of trash, it isn't slow and it is up-to-date.

### Steps to reproduce the unexpected behavior.

1. Enable Vitals in GNOME Extensions
2. Open Vitals settings and set the interval to check for changes to anything you want (for example to 3 or 5 seconds so you can see the issue well
3. Move your cursor. You'll notice it will pause for a moment, then resume like nothing happened, then pause again. Consistently.

### What distribution and version of Linux are you using?

Fedora 36, Linux kernel 5.17.5-200.fc35

### What version of Gnome are you using?

42, shell 42.0-3.fc36

ondrejsource, 2022-05-18:
I will post any logs happily, I'm just not sure where to find the relevant ones

corecoding, 2022-06-14:
Make sure the "Seconds between updates" is not too fast - ideally it shouldn't be much smaller than 5 seconds.

You can also try turning off Temperature, Voltage, Fan one at a time, you'll likely find out which group of sensors is causing the delay. What I've seen in the past is calls to read a file from /sys/class/hwmon causes the system lag.

I don't have any systems that do this, so creating a fix is going to be very difficult. Where I would start is by modifying sensors.js to remember how long each poll takes to return. It should be possible to say, for example, if a sensor takes longer than say, 1 second to poll, then it could be causing a delay. Key words here are 'could' as there is no way for Vitals to know if it is causing a delay.

I'm definitely open to merging a patch if you can help create a solution.

corecoding, 2022-06-21:
Please checkout the develop branch and see if it is any better please.
https://github.com/corecoding/Vitals#beta-testing

corecoding, 2022-06-21:
Making note - likely a duplicate of https://github.com/corecoding/Vitals/issues/250

XternalSoft, 2022-06-23:
Hello,
With the beta version, it seems to be better

corecoding, 2022-06-25:
That is good, thank you!

heynnema, 2022-07-04:
@corecoding After seeing the fan problem, I didn't test it long enough to see about the stack trace crashes.

ondrejsource, 2022-07-07:
Sorry for not responding for so long. Unfortunately, I don't use GNOME anymore, and I don't think I'll return anytime soon.

corecoding, 2022-07-10:
Thank you for replying. For others in this post please see #250.

commit b871acd0 (2022-06-26): attempt to fix #260
closed 2022-07-10 by corecoding